tipam/pi3d #176: 3d Apps for pi and web

mikolaskova, 2015-06-02:
Hello,

I want to build a 3d app that works on the raspberry and in a webbrowser- is it complicated to change p3d apps to something that can be shown in a browser without plugin?

Thanks and regards

Adriana

rec, 2015-06-02:
pi3d works on Python only and you can't run Python inside a browser.

closed 2015-06-02 by rec

mikolaskova, 2015-06-02:
yes, thank you. this I know. but maybe there is a known way how to port the p3d apps to javascript/WebGL f.e.?

rec, 2015-06-02:
Have you considered instead using http://threejs.org/?

mikolaskova, 2015-06-02:
yes...but this does not work on the raspberry...(WebGL does not work on raspberry, i tried in chromium, ice weasel...and read that WebGL and Rasberery does not work at all, only OpenGL ES...actually I don't understand much about the differences)...I need something that works both on a raspberry- anyhow... as standalone, in browser- and that is easy to port to something that works over the web...

I have an architectural model and the user should be able to switch on/of the lights and navigate around... there will be many stations so I want to use raspberries (and old monitors)...the same thing I want to realize on a website...(I already have with X3dom, that does not work on the raspberry, same as three.js wouldn't because it is WebGL)

rec, 2015-06-02:
https://gamedev.stackexchange.com/questions/150/what-is-the-difference-between-opengl-es-and-opengl

"Porting" pi3d to Javascript would be a huge amount of work, unfortunately.

paddywwoof, 2015-06-02:
At one level I would expect the WebGL stuff to migrate to the Raspberry Pi reasonably easily (for a.n.other expert in the machinery of javascript, not me.) as it's all geared "down" to OpenGL ES 2.0 and has to use ANGLE or mesa emulators, presumably so it runs on mobile browsers. 

You could risk doing everything in threejs and hope/lobby for someone to port it to a RPi compatibly browser! But possibly severe speed issues...

mikolaskova, 2015-06-02:
thank you, so I will try both.