tipam/pi3d #177: libEGL warning: DRI2: failed to authenticate

mikolaskova, 2015-06-02:
Hello,
I installed pi3d and downloaded the demos. I am trying to run them on my Kubuntu but in most of the cases there is only a black window and the message "libEGL warning: DRI2: failed to authenticate
" in the Terminal. What can I do to be able to run the examples?
Thanks and regards
Adriana

rec, 2015-06-03:
See answer 7 in the FAQs!  It's a problem with Raspbian...

https://pi3d.github.io/html/FAQ.html

closed 2015-06-03 by rec

mikolaskova, 2015-06-03:
thank you...but i'm trying to run the demos on kubuntu (desktop computer) not Rasbpian. I followed the instructions in the readme to  install it on the desktop computer, nevertheless I did also the things mentioned in FAQ 7... which results in the system not finding any libEGL anymore and error messages are different.
When running Snake.py: 
Traceback (most recent call last):
  File "Snake.py", line 24, in <module>
    DISPLAY = pi3d.Display.create(x=100, y=100, frames_per_second=10)
  File "/usr/local/lib/python2.7/dist-packages/pi3d/Display.py", line 517, in create
    display.opengl.create_display(x, y, w, h, depth=depth, samples=samples)
  File "/usr/local/lib/python2.7/dist-packages/pi3d/util/DisplayOpenGL.py", line 45, in create_display
    self.display = openegl.eglGetDisplay(EGL_DEFAULT_DISPLAY)
AttributeError: 'NoneType' object has no attribute 'eglGetDisplay'

paddywwoof, 2015-06-03:
Adriana, as Tom says the DRI2 error is one which cropped up before when another app had installed mesa on the Raspberry Pi (erroneously, as I don't think there's a version that uses the proprietory broadcom libraries) So this looks to be the most likely cause of your first error: that your hardware doesn't support DRI2 (do WebGL pages run OK), or that mesa is providing the wrong libraries (what do you get with $ sudo find / -name libEGL*).

Your second error looks to be due to not loading libEGL.so at all. The suggestion on the FAQ to get rid of any copies that are not in /opt/vc/lib is RPi specific and will not work on ubuntu as that will leave none! On this machine the working ones are in  /usr/lib/x86_64-linux-gnu/mesa-egl/

mikolaskova, 2015-06-03:
find / -name libEGL*

mikolaskova, 2015-06-03:
I'm working on kubuntu, no vmware. After I tried to set the symbolic links , I don't find anything anymore exept the libEGL.dll from wine installation, when using find / -name libEGL*

only /home/addi/.local/share/wineprefixes/firefox5/drive_c/Program Files (x86)/Mozilla Firefox/libEGL.dll

Also my GUI did not work propper anymore- probably i deleted the wrong files...so I had to installe kubuntu destkop again...)

I could install mesa-utils-extra, but there is no use of it, I suppose, when the libEGL missing at all.
I don't know, hot to get them again. In the mentioned folder ( /usr/lib/x86_64-linux-gnu/mesa-egl/) there are now the files as on the image below
[image]

(on my raspberry i could install the pi3d and it works great!! But to develop I would prefer to use my desktop, so thats why  I really would like to be able to install it)

paddywwoof, 2015-06-03:
On my laptop (ubuntu 14.04LTS) mesa-utils-extra installed libEGL.so.1.0.0 in the same directory as libGLESv2.so.2.0.0 along with symbolic links libEGL.so.1 and libGLESv2.so.2 in the same directory pointing to these. 

I think you will need to re-install the mesa libraries again to get pi3d working. If you do, see if there's an error or conflicting something message.

The point about the desktop sounds like there are some kubuntu libraries that might be conflicting. I don't know why mesa-utils-extra would behave differently with KDE compared with GNOME but there might be something. I thought at one stage, on one machine I tried, that there were ubuntu versions of libEGL and libGLESv2 but they don't seem to be on this laptop. Did you know which files you deleted? Were they just symbolic links or real files? How many?

If you do a "find" after re-installing the kubuntu desktop are there any other copies of libEGL?

The dll files in wine/firefox are not going to work from linux (even under windows, with python 64 bit I had to recompile the ANGLE dlls)

mikolaskova, 2015-06-03:
thank you very much, I tried to install the mesa libraries, but  there seems to be something very messed up... I get things like E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages....

I think I just set the symbolic links, but wrong (becaus it was for raspbian- i realized later)... and I don't know how to resolve this chaos now..

find can't find the libEGL even after reinstalling kubuntu desktop

so I will try to install a new system in VirtualBox

paddywwoof, 2015-06-03:
I can't see why the files you found the first time with "find" are not still there when you re-run "find", but now symbolic links pointing to a non-existent location.

maybe package broken by re-installing desktop. Is this any help
http://askubuntu.com/questions/124845/eerror-pkgproblemresolverresolve-generated-breaks-this-may-be-caused-by-hel

Is the VirtualBox in windows? If so you could install pi3d in that directly.

WayneKeenan, 2018-04-02:
I  just read the FAQ and would like to highlight that this is occurring on 2018-03-13 Stretch RPi3+:

```shell
pi@pivr:~/local.projects $ sudo find / -name libEGL*
/usr/lib/arm-linux-gnueabihf/libEGL.so.1
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0
/usr/lib/chromium-browser/libEGL.so
find: ‘/run/user/1000/gvfs’: Permission denied
/opt/vc/lib/libEGL_static.a
/opt/minecraft-pi/lib/brcm/libEGL.so
/opt/minecraft-pi/lib/mesa/libEGL.so
pi@pivr:~/local.projects $ sudo find / -name libGLESv2*
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2
/usr/lib/chromium-browser/libGLESv2.so
find: ‘/run/user/1000/gvfs’: Permission denied
/opt/vc/lib/libGLESv2_static.a
/opt/minecraft-pi/lib/brcm/libGLESv2.so
/opt/minecraft-pi/lib/mesa/libGLESv2.so
```

I read in the FAQ entry:

> This issue is being looked into by the maintainers of Raspbian so, hopefully, it will be fixed in later releases.`  

I wondered if you've had an any update from the Raspbian maintainers on this?  

Thanks
Wayne

paddywwoof, 2018-04-02:
Wayne, Thanks for the heads-up. I will have a play with the new OS on a new SD, though I have apt-get update/upgraded and it still works ok. Is your install completely clean or might there be something else that could have installed the mesa stuff?. Maybe they've moved the so files (again) or maybe the OpenGL drivers are now fully working and not switched off by default. In which case, maybe I ought to see if pi3d can use them instead. I suspect they will be slower on OpenGLES systems like the RPi but it would be quicker on computer sized computers. Will let you know what I find.

WayneKeenan, 2018-04-02:
Hi Paddy,   it was a 'fresh 20180-3-13' SD card and hour or so before I commented, but, I had been installing and building stuff, e.g. OpenHMD Deb packages and also from source, also GTK-JSTest amongnst a handful of others so there's a chance one of them did pull it in.  
I can check the  OpenHMD deb deps and for the source build I specifically didn't install or enable the OpenGL based test.  

Something has changed though, when I used to install VRZero on Jessie it was necessary to disable `hciattach` service otherwise OpenGL init failed, but I hadn't done it for Stretch this time round and it worked regardless, so, perhaps it may also be as you suspect; the default OpenGL driver may have changed.

I need to test my install instructions from scratch for my project so will have another go later today.

paddywwoof, 2018-04-02:
OK, I need to do a test on a new card (which I ought to have done before the v2.23 issue really) also an ubuntu system. It's not reasonable to expect everything else to stand still I suppose!

paddywwoof, 2018-04-05:
@WayneKeenan, sorry not to get back sooner - I've been away for a couple of days. I can't reproduce the error with a fresh install (just pip install pi3d and apt-get udpate, upgrade) so I think it's definitely one of the other things you've installed.

pi3d.__init__() basically checks to see if libbcm_host.so can be found (using ctypes.util.find_library()) if so works on the basis it's a RPi and then looks in /opt/vc/lib and /usr/lib for different named versions of libEGL and libGLESv2 you could try manually doing the same thing and seeing at what point it diverges from the expected trajectory.
Paddy

cescgie, 2018-12-19:
It seems that this issue has not been resolved yet. Or someone has an alternative?

paddywwoof, 2018-12-19:
@cescgie some of the causes of this error have been resolved but others may not.. what are you running to get the error and on what machine?

wambiditu, 2019-02-24:
While I'm not yet a Pi3D user, I have encountered this error on a Raspberry Pi SD card that is just a day old.  I think the significant changes that I made to my Pi are (a) installing XScreenSaver GL and (b) installing PyQt5, both from Raspbian repository.

The screensaver works well enough for me (but slowly).

But when I try to execute a Python script that uses Qt, I get the message "libEGL warning: DRI2: failed to authenticate"

`lsb_release -a` says I am using Raspbian 9.8 (stretch).
I am using Python 3.5.3
QT_VERSION_STR is 5.7.1

paddywwoof, 2019-02-24:
Hi thanks for info. I've not used pyqt5 on the raspberry pi yet (but I think I've used pyqt4 ok) Does it use the OpenGLES driver or the new GL driver set up in raspi-config? What happens if you pip3 install pi3d then open a python3 terminal and type

import pi3d
d = pi3d.Display.create(w=100, h=100)

Les-A, 2019-03-11:
I hope you don't mind me adding to this. I run a Pi 3 and rebuilt it yesterday and have the same problem (message "libEGL warning: DRI2: failed to authenticate") but with HP-LIP. uname -a output is:
Linux rpi3 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
I tried following instructions pointed to above (blindly) to create some links and I guess its now broke so will have to rebuild for safety but output was as follows:

sudo find / -name libEGL*
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0
/usr/lib/arm-linux-gnueabihf/libEGL.so.1
find: ‘/proc/769/task/769/net’: Invalid argument
find: ‘/proc/769/net’: Invalid argument
find: ‘/run/user/1000/gvfs’: Permission denied
/opt/vc/lib/libEGL_static.a

pi@rpi3:~ $ sudo find / -name libGLESv2*
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0
find: ‘/proc/769/task/769/net’: Invalid argument
find: ‘/proc/769/net’: Invalid argument
find: ‘/run/user/1000/gvfs’: Permission denied
/opt/vc/lib/libGLESv2_static.a

This is mainly for information but any pointers as to how to really put it right would be appreciated. If the situation changes after the rebuild I will post it here.

N2Network, 2019-07-25:
Was anyone able to resolve this on the Pi? I'm running DebianBuster

pi@raspberrypi:~ $ sudo find / -name libEGL*
find: ‘/proc/857/task/857/net’: Invalid argument
find: ‘/proc/857/net’: Invalid argument
/opt/vc/lib/libEGL_static.a
/opt/minecraft-pi/lib/mesa/libEGL.so
/opt/minecraft-pi/lib/brcm/libEGL.so
/usr/lib/chromium-browser/swiftshader/libEGL.so
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.1.0
/usr/lib/arm-linux-gnueabihf/libEGL.so
/usr/lib/arm-linux-gnueabihf/libEGL.so.1
/usr/lib/arm-linux-gnueabihf/libEGL_mesa.so.0
/usr/lib/arm-linux-gnueabihf/libEGL_mesa.so.0.0.0
find: ‘/run/user/1000/gvfs’: Permission denied
pi@raspberrypi:~ $ sudo find / -name libGLESv2*
find: ‘/proc/857/task/857/net’: Invalid argument
find: ‘/proc/857/net’: Invalid argument
/opt/vc/lib/libGLESv2_static.a
/opt/minecraft-pi/lib/mesa/libGLESv2.so
/opt/minecraft-pi/lib/brcm/libGLESv2.so
/usr/lib/chromium-browser/swiftshader/libGLESv2.so
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.1.0
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2
/usr/lib/arm-linux-gnueabihf/libGLESv2.so
find: ‘/run/user/1000/gvfs’: Permission denied

Odinnbh, 2020-01-01:
It seems to be a dead end but has anyone solved this? I'm having similar issues, trying to run JACK. When I run cjackctl it comes up with libEGL warning: DRI2: failed to authenticate.

pi@raspberrypi:~ sudo find / -name libEGL*
/opt/minecraft-pi/lib/brcm/libEGL.so
/opt/minecraft-pi/lib/mesa/libEGL.so
/opt/vc/lib/libEGL_static.a
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/lib/arm-linux-gnueabihf/libEGL.so.1.1.0
/usr/lib/arm-linux-gnueabihf/libEGL_mesa.so.0
/usr/lib/arm-linux-gnueabihf/libEGL.so
/usr/lib/arm-linux-gnueabihf/libEGL.so.1
/usr/lib/arm-linux-gnueabihf/libEGL_mesa.so.0.0.0
/usr/lib/chromium-browser/swiftshader/libEGL.so
/usr/lib/scratch3/libEGL.so
pi@raspberrypi:~ $ sudo find / -name libGLESv2*
/opt/minecraft-pi/lib/brcm/libGLESv2.so
/opt/minecraft-pi/lib/mesa/libGLESv2.so
/opt/vc/lib/libGLESv2_static.a
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2
/usr/lib/arm-linux-gnueabihf/libGLESv2.so
/usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.1.0
/usr/lib/chromium-browser/swiftshader/libGLESv2.so
/usr/lib/scratch3/libGLESv2.so

Any ideas or pointers in the right direction appreciated.

amazingfly, 2020-09-27:
I was having this issue on my RPI3B+ and was scouring the internets and nothing seemed to fit.  It was because I was using windows remote desktop to log in!  So maybe that was just me being a noob, but I have yet to see that as an answer anywhere I've looked.

So I enabled VNC on my RPI and installed VNC on my desktop.

Doesn't work over remote desktop.  But maybe thats

myfingerhurt, 2022-04-11:
Ubuntu 20.04 LTS RPI4B

I encountered this warning when I ran firefox which might be different to yours.

```
$ DISPLAY=:10.0 firefox -P Proxy
libEGL warning: DRI2: failed to authenticate
```

When I open a terminal, I saw an error. `-bash: /dev/null: Permission denied`
Then I did this:
```
ls -l /dev/null
rm /dev/null
mknod /dev/null c 1 3
chmod 666 /dev/null
ls -l /dev/null
```
After finishing those commands, `libEGL warning` seems gone.

Here is the original post.
https://unix.stackexchange.com/questions/146633/bash-dev-null-permission-denied