tipam/pi3d #178: .obj File exported from Blender not shown

mikolaskova, 2015-06-03:
Hello, 
I exported the blender default cube to .obj format (according faq instructions) and tried to display it in the Triceratops example (changed the path to my .obj File)... the screen remained black.

When replacing the original Triceratops with one of the other example-obj-Files (f.e. cottage) it worked fine. When importing the cottage-obj-File from your examples to blender, modifying and re-exporting it: worked fine.

Just my own blender model exports are not shown...

What could be wrong with my obj. file? (it's here http://www.mikolaskova.cz/downloads/meine/mein.obj) 

Thanks and Regards

Adriana

paddywwoof, 2015-06-03:
It looks ok to me (assuming you have an appropriate mein.mtl file with Material.001 defined) but it's very big. You would need to set the z value to 101.0 to see it at all and nearer 500.0 to see it as a cube.

mikolaskova, 2015-06-03:
I tried different scales...the default cube is very small...  the one I use now is around 100 http://www.mikolaskova.cz/downloads/meine/mein.obj 
there is this mtl-File http://www.mikolaskova.cz/downloads/meine/mein.mtl
the z-value in Blender is to 0-

paddywwoof, 2015-06-03:
Hmm, this has thrown up some interesting effects. The fundamental reason is (probably) that you are trying to use a shader that requires uv mapping (if it's TricerotopsModel.py it uses uv_light) but your model doesn't have any uv coordinates included. If, before you export from Blender, you tab to edit mode and key U then unwrap it should work fine.  Alternatively, if you will always use self colour with your model you can use a mat_light type shader. NB mat_bump and mat_relfect need uv mapping to work and will crash out similarly.

The curious thing is that if I create an object in Blender 120 or more triangles and export as obj without uv mapping then it still works ok, but with odd reflections. Less than that and it crashes horribly. The shader must be accessing memory that is getting filled up with something benign with the larger arrays.. or something.

closed 2015-06-03 by paddywwoof

mikolaskova, 2015-06-04:
Thank you very much- ich unwrapped and at least there is something similar to my model.
Now I'm confident, that somehow I will manage to transform my blender models in a way they can be used with pi3d.

PS- no I did not yet try everything you suggested, i will now