tipam/pi3d #180: Running any demo leads to AssertionError

qubodup, 2015-06-07:
Hi, when I try to run on Arch Linux 64bit with proprietary AMD drivers (did not test open drivers), I get AssertionError with all demos from http://github.com/pi3d/pi3d_demos

```
$ python2 Conway.py 
Traceback (most recent call last):
  File "Conway.py", line 14, in <module>
    DISPLAY = pi3d.Display.create(w=WIDTH, h=HEIGHT)
  File "/home/qubodup/pi3d_demos-master/pi3d/Display.py", line 522, in create
    display.opengl.create_display(x, y, w, h, depth=depth, samples=samples)
  File "/home/qubodup/pi3d_demos-master/pi3d/util/DisplayOpenGL.py", line 70, in create_display
    assert self.context != EGL_NO_CONTEXT
AssertionError
```

My system:
```
Distro        4.0.4-2-ARCH x86_64
Mainboard     FOXCONN A7GM-S 2.0
CPU           AMD Phenom(tm) 9550 Quad-Core Processor
              4 x 2200 MHz, 64-bit
RAM           7.8G
Video         Advanced Micro Devices, Inc. [AMD/ATI]
              Cape Verde PRO [Radeon HD 7750 / R7 250E]
OpenGL        4.4.13374 Core Profile Context 15.20.1013, GLSL 4.40
GCC           gcc (GCC) 5.1.0
```

paddywwoof, 2015-06-07:
Hi, if it gets to that assertion than pi3d has managed to load a version of libEGL.so but it's not one that cuts the mustard, for some reason. I've not set up pi3d with Arch (apart from on the RPi) so I don't know what the relevant ``mesa-utils-extra`` looks like. When I tried to set up on a virtual machine with Debian I struggled to get the mesa shared library files set up, though it was a doddle with ubuntu and lubuntu.

On this (ubuntu) laptop I get::

    $ sudo find /usr/lib/ -name libEGL* -ls # or /lib/ or /usr/local/lib/ takes a while for find /
    8527809    0 lrwxrwxrwx   1 root     root           15 Mar 11 22:47 /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1 -> libEGL.so.1.0.0
    8522132  152 -rw-r--r--   1 root     root       151824 Mar 11 22:47 /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1.0.0

What do you get on your setup? In the last version (v2.4) I have had to change the location to find the libraries on the RPi (done in pi3d/constants/__init__.py) because raspbian puts them in /opt/vc/lib/ and ubuntu Mate puts them /usr/lib/ but also has mesa versions which override them!

qubodup, 2015-06-08:
I have `/usr/lib/mesa/libGLESv2.so` owned by mesa and `/usr/lib/libGLESv2.so` owned by catalyst-libgl.

The error is the same, no matter whether I use the `/usr/lib/mesa/` or `/usr/lib/` libraries.

paddywwoof, 2015-06-08:
Hard to know what the issue is. It might be possible to gain a little more info using an x11 version of the test program I used to debug someone's windows setup. See attached - oops can't attach... I will put it here: http://www.eldwick.org.uk/files/test_egl_x11.py
You will need to edit the path and names for the .so files (also if you git cloned pi3d rather than pip installed it you will need to uncomment and alter sys.path lines at start)

paddywwoof, 2015-06-16:
Did you run the debug app? Any info?

closed 2015-06-16 by paddywwoof

qubodup, 2015-06-16:
With 
```path = "/usr/lib/mesa"```
or
```path = "/usr/lib"```
I get this output:
```$ python3 test_egl_x11.py 
For ref
  EGL_FALSE=0x0000
  EGL_BAD_DISPLAY=0x3008
  EGL_BAD_ATTRIBUTE=0x3004
  EGL_NOT_INITIALIZED=0x3001
  EGL_BAD_PARAMETER=0x300C
  EGL_WINDOW_BIT=0x0004

Traceback (most recent call last):
  File "test_egl_x11.py", line 50, in <module>
    assert r == EGL_TRUE             #<<<<<<<<<<<<<<<<<<<<<<
AssertionError
$ python2 test_egl_x11.py 
For ref
  EGL_FALSE=0x0000
  EGL_BAD_DISPLAY=0x3008
  EGL_BAD_ATTRIBUTE=0x3004
  EGL_NOT_INITIALIZED=0x3001
  EGL_BAD_PARAMETER=0x300C
  EGL_WINDOW_BIT=0x0004

Traceback (most recent call last):
  File "test_egl_x11.py", line 50, in <module>
    assert r == EGL_TRUE             #<<<<<<<<<<<<<<<<<<<<<<
AssertionError```

paddywwoof, 2015-06-16:
Thanks, it looks to be failing eglInitialize() which is where it failed on windows when it didn't load the correct dll version. I have copied the so files from my computer here http://eldwick.org.uk/files/libegltest.tar.gz You could try putting them somewhere on your system and changing the path (and probably file names) in test_egl_x11.py Not sure how much further forward I will be if they work or if they don't but it's all additional info!

qubodup, 2015-06-16:
I had to change path and make symbolic links libEGL.so.1 and libGLESv2.so.2. The result was:
<pre>$ python2 test_egl_x11.py 
For ref
  EGL_FALSE=0x0000
  EGL_BAD_DISPLAY=0x3008
  EGL_BAD_ATTRIBUTE=0x3004
  EGL_NOT_INITIALIZED=0x3001
  EGL_BAD_PARAMETER=0x300C
  EGL_WINDOW_BIT=0x0004

Traceback (most recent call last):
  File "test_egl_x11.py", line 50, in <module>
    assert r == EGL_TRUE             #<<<<<<<<<<<<<<<<<<<<<<
AssertionError</pre>

paddywwoof, 2015-06-16:
Yes, sorry about that. Using .so files from ubuntu was never going to work, looking at it now I see there are 20 shared object dependencies needed for libegl and libglesv2.

There is obviously a conflict somewhere. Searching on line there seem to be a few instances where another application stops the catalyst drivers from working and removing ati, mesa etc drivers is required.