tipam/pi3d #182: Use pygame where available

stuaxo, 2015-07-11:
I made a branch to use pygame when available ..
https://github.com/stuaxo/pi3d/tree/pygame_where_available

It's bit hacky at the moment, but is enough for me to run demos that need the mouse in Ubuntu.
The display doesn't seem setup properly, as the forest-walk demo just displays in the top left corner.
http://imgur.com/VksgcaK

For this to be less hacky, the concept of platforms probably needs to change, maybe pygame should be a platform, or there could be constants for USE_X11.

Also, all (most) of classes _win could be _pygame .. or maybe extend pygame ones?

There should be a way of toggling pygame on / off for testing.

Using pygame is nicer (for me) than setting permissions on /dev/input/mice ... on the other hand it can be a bit of a pain using pygame in a virtualenv - though I have a workaround for that.

Let me know if this seems useful, if so I can probably clean it up.

paddywwoof, 2015-07-12:
Stuart, I'm back now so will have a more thorough look at this. After a quick glance what you've done looks excellent, and very much along the lines I was thinking.

Because the platform setting and loading the shared libraries happens in constants.__init__.py (i.e. at import pi3d) I think the pygame options has to be slightly different from the other platform constants. I think there probably should be parameter, maybe passed to Display.create() that switches pygame on and if PLATFORM has been determined to be windows this would be set automatically. Then, as you say, all the subsequent windows conditional sections become pygame conditional.

I did have a half hearted test to see if pygame would work on the Raspberry Pi to provide the drawing surface (and had tried using x11 previously) without any success, but I would have thought it must be possible!

So, yes, if you've got the time to tidy it up that would be great, otherwise let me know and I can work from here.

stuaxo, 2015-07-13:
I'm using pygame to run this on Linux, so I can run this as a normal user in a virtualenv, so maybe it makes sense to have an extra concept (backend, implementation or feature maybe?), as PLATFORM_PYGAME seems a bit of an odd one out.

I've missed a couple of checks around the keyboard as these ones don't work:
DogFight.py, Gui.py,  PictureFrame.py

I'll try and get the example that doesn't work, working + see if I can come up with something that makes sense.

There are some instructions for getting pygame on pi working here
https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=33157&p=332140&hilit=croston%2bpygame#p284266

Not sure if that helps ?

stuaxo, 2015-07-13:
BTW - Pygame is still a little bit of a pain to use in a virtualenv, I've made 'vext.pygame' to make workaround this, until they sort this out.

paddywwoof, 2015-07-13:
Yes I don't think pygame really is a platform in the same sense. The PLATFORM 'global static variable' is use for two related but different things 1. in constants.__init__.py it determines whereabouts to load the libEGL and libGLESv2 libraries 2. in DisplayOpenGL.py (but for the Raspberry Pi also partly in constants.__initi__.py) it is used to obtain the window and surface to use with the egl function calls. The Keyboard and Mouse classes then have to use whichever windowing system is in play.

So, as you say, using pygame should be an additional option but one that selected automatically with windows and, probably, not available on Raspberry Pi or Android (which, ironically already uses pygame but via the internals of pythonforandroid)

DogFight and Gui use the events system that accesses /dev/input/ directly and really only comes into it's own on the Raspberry Pi where you might want to make a game console. i.e. don't spend any time trying to get them to work on linux or windows etc.

PS: pygame comes re-installed with raspbian so that's not a problem, but when I modified the code so that DisplayOpenGL tried to obtain the window and surface via pygame I failed to get it to work. It may have been a simple mistake, I didn't persist for very long!

stuaxo, 2015-07-13:
I'm having a look at pi3d (and some other stuff) to see if it's what I want to build a music vis in, so getting input working would be handy..

I have a raspberry pi somewhere, so in theory could test stuff, it's a been a while though.    

Would be nice if there was a solution that would work cross platform (pi, pc, and android) and install via pip ... closest I can see at the moment is pyglet and pyget android, but may well be something else.

paddywwoof, 2015-07-13:
I did something involving music vis in a google competition https://devart.withgoogle.com/#/project/17120531 - If you end up doing most of your 'fancy' graphics in shaders then I don't suppose it matters too much what the front end is made from (apart from python being easier to get code working generally).

Have you looked at kivy? I think the input stuff of pygame should be good for most situations. Android (and iOs) will always be a bit different - touch and accelerometer etc.

stuaxo, 2015-07-18:
That music vis looks really interesting, will definitely have a look .. 
I don't have too much time in the next couple weeks... so probably won't be able to look at this quite yet.

Kivy looks really interesting, I've been a bit lazy in actually learning it, the cross platform bit looks really good.

paddywwoof, 2015-08-14:
Option now added to Display.create(), thanks for input.

closed 2015-08-14 by paddywwoof

stuaxo, 2015-08-14:
Cheers :)