tipam/pi3d #187: PyPi 2.9 package doesn't install correctly

swehner, 2016-01-17:
I've tried to install the v2.9 release from pypi and it gives the following error:
```
...
byte-compiling /usr/local/lib/python3.4/dist-packages/six.py to six.cpython-34.pyc

running install_data

error: can't copy 'ChangeLog.txt': doesn't exist or not a regular file
```

Seems like the tar https://pypi.python.org/packages/source/p/pi3d/pi3d-2.9.tar.gz is really missing the ChangeLog.txt file that was there in previous versions...

paddywwoof, 2016-01-17:
@swehner hmm, I think this is something to do with the change https://github.com/tipam/pi3d/pull/183 which I have now commented out and re-uploaded a revised tar.gz I've installed it on this computer and it seemed to go ok. Could you check and let me know.

Thanks for checking and bringing to my attention.

swehner, 2016-01-17:
Yep - Works fine now! Thanks!

closed 2016-01-17 by swehner