tipam/pi3d #191: Camera model_view not updating

John-P, 2016-02-01:
When I relocate the camera with `camera.reset()` and `camera.relocate()`, `camera.model_view` does not update. I was using this to calculate screen coordinates from a 3D point.

```
def screenXY(x, y, z):
    point = np.dot(np.array([x, y, z, 1]), camera.model_view)

    screenx = round(((point[0] / point[3] / 2.0) + 0.5) * display.width)
    screeny = round(((point[1] / point[3] / 2.0) + 0.5) * display.height)
    return screenx, screeny, 0
```

John-P, 2016-02-01:
Solved. I should have been using camera.mtrx

paddywwoof, 2016-02-01:
@John-P There is a slightly illogical difference between the way the Camera matrix stuff is done c.f. the Shape matrices. For the latter all the components (rotation, translation, scale etc) are held and only multiplied together in Shape.draw() but for the Camera the multiplication is done when the relevant method is called. The reasoning was (probably) that the same model_view matrix could be used for drawing lots of shapes but, in retrospect, it would have been better to do the multiplication as part of Shape.draw() and set the camera_moved flag to prevent it being recalculated.

Does this mean that you have got pi3d working, after a fashion, on the mac?

John-P, 2016-02-02:
@paddywwoof Unfortunately not yet working on os x. I am using it on linux for now. I have been a little busy and not progressed much on the mac front. It looks like the mesa recipe in homebrew does not install the EGL/GL ES drivers so will have to do a manual install and try again.

John-P, 2016-02-02:
@paddywwoof  As a side note. Being able to access a camera's scale would be useful to be able to scale movements in the screen coordinates to the camera scale. This could be done by looking at the `camera.mtrx` but is a bit complicated for simple cases.

paddywwoof, 2016-02-02:
@John-P Good idea. I've added added self.scale = scale to Camera.__init__() and reset()

closed 2016-02-02 by paddywwoof