tipam/pi3d #192: Best way to write to the stencil buffer

John-P, 2016-02-09:
I am trying to achieve an outline effect such as in [this example](http://www.flipcode.com/archives/Object_Outlining.shtml). I'm not very familiar with the internals of pi3d or opengl in general but have tried this so far:

```
    pi3d.opengles.glClearStencil(0)
    pi3d.opengles.glClear(GL_STENCIL_BUFFER_BIT)

    pi3d.opengles.glEnable(GL_STENCIL_TEST)
    pi3d.opengles.glDisable(GL_DEPTH_TEST)

    pi3d.opengles.glStencilFunc(GL_ALWAYS, 1, -1)
    pi3d.opengles.glStencilOp(GL_KEEP, GL_KEEP, GL_REPLACE)

    testcube.draw(shader=meshblue)

    pi3d.opengles.glStencilFunc(GL_NOTEQUAL, 1, -1)
    pi3d.opengles.glStencilOp(GL_KEEP, GL_KEEP, GL_REPLACE)

    testcubecopy.set_line_width(3.0)
    #pi3d.opengles.glLineWidth(3)
    #pi3d.opengles.glPolygonMode(GL_FRONT, GL_LINES)
    #glPolygonMode not supported in GL ES

    testcubecopy.draw(shader=outline)
```

paddywwoof, 2016-02-09:
Hi, no, that works really well. It's hard to tell what's not right without seeing the full code but this runs ok for me (linux laptop, not tried on RPi). I've used a MergeShape to show the kind of issue you can get with set_line_width() method of switching an arbitrary shape to GL_LINES GL_LINE_STRIP or GL_LINE_LOOP. Sometimes it's better setting the argument stip=False but this can leave 'gaps' in the mesh, probably not as significant for an outline.
```
#!/usr/bin/python
from __future__ import absolute_import, division, print_function, unicode_literals

import demo
import pi3d

DISPLAY = pi3d.Display.create(x=10, y=10, w=900, h=600, frames_per_second=25)
DISPLAY.set_background(0.4, 0.6, 0.8, 1.0)      # r,g,b,alpha

CAM = pi3d.Camera() # default instance camera perspecive view

shader = pi3d.Shader("uv_light")
flatsh = pi3d.Shader("mat_flat")

OUTLINE = [1.0, 0.0, 0.0] # colour for outline

#Create textures
shapeimg = pi3d.Texture("textures/straw1.jpg")

#Create shape
shape = pi3d.MergeShape()
asphere = pi3d.Sphere(sides=16, slices=16)
shape.radialCopy(asphere, step=72)
shape.position(0.0, 0.0, 5.0)
shape.set_draw_details(shader, [shapeimg])

# Display scene and rotate shape
while DISPLAY.loop_running():
  pi3d.opengles.glClearStencil(0)
  pi3d.opengles.glClear(pi3d.GL_STENCIL_BUFFER_BIT)

  pi3d.opengles.glEnable(pi3d.GL_STENCIL_TEST)
  pi3d.opengles.glDisable(pi3d.GL_DEPTH_TEST)

  pi3d.opengles.glStencilFunc(pi3d.GL_ALWAYS, 1, -1)
  pi3d.opengles.glStencilOp(pi3d.GL_KEEP, pi3d.GL_KEEP, pi3d.GL_REPLACE)

  shape.set_material((0.5, 0.5, 0.5)) # otherwise will have a tinge
  shape.set_line_width(0.0) # resets to GL_TRIANGLES
  shape.draw(shader=shader) # draw solid version

  pi3d.opengles.glStencilFunc(pi3d.GL_NOTEQUAL, 1, -1)
  pi3d.opengles.glStencilOp(pi3d.GL_KEEP, pi3d.GL_KEEP, pi3d.GL_REPLACE)

  shape.set_material(OUTLINE)
  shape.set_line_width(3.0) # try with , strip=False
  shape.draw(shader=flatsh)

  shape.rotateIncY(1.247)
  shape.rotateIncX(0.1613)
```

John-P, 2016-02-09:
I'm trying to mask a shape using the stencil buffer but cannot get it to work. Tried running the example above and can't see any masking.

paddywwoof, 2016-02-09:
That sounds odd. I wonder what the difference is - maybe it's in the graphic card or emulator. What are you running on. This is what it looks like on this laptop (64bit lenovo i5 sandybridge ubuntu 14.04) https://drive.google.com/file/d/0B0RYleARqaxsNWoxZG4yOFlkTkE/view?usp=sharing

John-P, 2016-02-10:
I'm running on a raspberry pi 2 running raspian lite jessie. Mine looks like this except all of the edges (all lines between vertices) show as red. Tried to take a screenshot with scrot but the window does not show up so here is a photo:

https://dl.dropboxusercontent.com/u/7785550/12714515_1245750358771770_887889542_n.jpg

John-P, 2016-02-10:
Just tried on my ubuntu machine ans is the same. I commented out the 'import demo' because it didn't look like it was being used an caused an import error. Is something from there needed maybe?

https://dl.dropboxusercontent.com/u/7785550/Screenshot%20from%202016-02-10%2010%3A56%3A21.png

paddywwoof, 2016-02-10:
OK, sorry about this. I hadn't realized that GL_STENCIL_SIZE was a variable capability and needs to be specifically asked for when generating the drawing context. If not specified it probably produces random results. On this GPU and the Raspberry Pi it seems to be there as 0 or 8 so I've modified pi3d/util/DisplayOpenGL.py to
```
    attribute_list = c_ints((EGL_RED_SIZE, 8,
                             EGL_GREEN_SIZE, 8,
                             EGL_BLUE_SIZE, 8,
                             EGL_DEPTH_SIZE, depth,
                             EGL_ALPHA_SIZE, 8,
                             EGL_BUFFER_SIZE, 32,
                             EGL_SAMPLES, samples,
                             EGL_STENCIL_SIZE, 8,
                             EGL_SURFACE_TYPE, EGL_WINDOW_BIT,
                             EGL_NONE))
```
The ``import demo`` is just a convenience for
```
import sys
sys.path.insert(1, '/home/pi/pi3d') 
```
so you can clone the github version and use develop even if you have installed the current live version with ``sudo pip install pi3d``

In the demo code I posted earlier I found that the lines disabling depth test and the repeated glStencilOp weren't needed.

Finally. You can do a similar (but not as nice) outlining by moving an object away from the Camera, scaling it up, drawing as coloured lines (as in the stencil method), then moving it back and resetting scale. There is a convenience method Camera.get_direction() that returns a vector you can use.

closed 2016-02-10 by paddywwoof

John-P, 2016-02-10:
Thanks for this. Working great now!