tipam/pi3d #193: 2d Alpha issues/question

swehner, 2016-02-29:
Hi,

I've played a bit with the Minimal_2d example together with a solid bg and a semitransparent ImageSprite in front of it running it directly on the Raspberry pi from the console (no X11).
What happens is, that at the borders of the transparent shape, the console framebuffer is visible...

This is the code:
https://gist.github.com/swehner/aa8376d24dee07e14578

This is what a screenshot shows:

[image]

And this is a blown up photo of one the borders - if you look closely you can see the console text behind it:

[image]

I'm not sure if I'm doing something wrong, or if this is expected behavior.
Thanks for any insight you can give me on this!

paddywwoof, 2016-02-29:
Well I had noticed this before but not actually figured it out. Now I have I'm not entirely sure what's the best thing to do. Essentially the blend function defined in pi3d/pi3d/util/DisplayOpenGL.py line 88 is
glBlendFunc(GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA)
which mixes the new pixels with the ones already drawn linearly according the the alpha values of the new pixels. Unfortunately it also blends the alpha values so if you blend
(1.0, 0.0, 0.0, 1.0) with (0.5, 0.5, 0.5, 0.5) you end up with (0.75, 0.25, 0.25, 0.75) i.e. partially transparent.

I will have a play with the options but I think it would be necessary to switch to
https://www.khronos.org/opengles/sdk/docs/man/xhtml/glBlendFuncSeparate.xml
The alternative, of course would be to set the background alpha to 1.0

paddywwoof, 2016-02-29:
opengles.glBlendFuncSeparate(GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA, 
                                 GL_ONE_MINUS_DST_ALPHA, GL_DST_ALPHA)

seems to work ok. Need to think if this is always going to be suitable

swehner, 2016-03-01:
Ok - thanks for the info - i hadn't looked into that...
In our case the background has to be set transparent at some point because we want to play a video under the OpenGL layer.
So I see two options that should work:
 * Use Display.set_background to set it transparent only when the video is actually being played (the rest of the time it should all work fine if we set alpha to 1)
 * Change the blend functions the way you've done here from our app after initializing the display...

So in any case no change would be needed from pi3d... What we're doing isn't a very common case, so I don't think it's worth changing the "default" setting that works fine for almost all cases.
Thanks a lot for the info!!!

closed 2016-03-01 by swehner

swehner, 2016-03-08:
Just FYI: in the end we ended up using
`glBlendFuncSeparate(GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA, 1, GL_ONE_MINUS_SRC_ALPHA)`
This way the blending works as we expected - We expect the blended result to be the following:
```
(1 - Ad) = (1 - As) * (1 - Ad)
Ad = Ad + As - Ad*As
```
and using these values with the opengl blending formula we obtain the same result
```
Ad = As * 1 + Ad * (1 - As)
Ad = As + Ad - Ad*As
```
Thanks again!

paddywwoof, 2016-03-08:
@swehner I think I will change pi3d default as per your version.

I can't think of any situation where blending something on top of an opaque object and having that object become partially transparent mid blend would be expected behaviour.