tipam/pi3d #195: StereoCamera - Only Background

cowminer27, 2016-06-18:
when using the stereo camera, i always get a blank background.

this happens with my own code or the ForestStereo.py example.
I am running it on Python3, Debian Jessie, Raspberry Pi 1B

paddywwoof, 2016-06-19:
Hi, thanks for highlighting this.  I'm away camping now so can't check anything. I will have a look when I get back on Tuesday. I will put that demo into the standard tests as it probably uses some functionality not used elsewhere.

cowminer27, 2016-06-20:
thx so much!

paddywwoof, 2016-06-21:
Little bit harder to find than I thought. The problem was due to the modified blend function (see commit 898bca0 18th Apr 16) and I will push a mod to pi3d/util/StereoCam to the effect

    opengles.glColorMask(1, 1, 1, 1)

as the OffscreenTexture alpha seems to default to 0

In the mean time you can get the demo to work by changing the background alpha to less than 1.0 ie

    DISPLAY = pi3d.Display.create(x=200, y=200)
    DISPLAY.set_background(0.4, 0.8, 0.8, 0.99)      # r,g,b,alpha

cowminer27, 2016-06-21:
Thanks so much!

closed 2016-10-24 by paddywwoof