tipam/pi3d #197: Dispmanx alpha issues

swehner, 2016-08-15:
Hi there,

I've been trying to get a transparent OpenGL layer on top of a solid dispmanx layer and to get the alpha composition right. From what I've seen I think that the Dispmanx element containing the OpenGL render surface should be marked with the DISPMANX_FLAGS_ALPHA_PREMULT flag instead of the default settings.

I've made a small demo here: https://github.com/swehner/pi3d_demos/blob/test_dispmanx_alpha/alpha_test.py
You can check it out from the test_dispmanx_alpha branch and run it on the raspberry pi.

You can run several combinations of the program:
 * Drawing opaque background and transparent shapes in just OpenGL (alpha_test.py) (this will be the "gold standard" to compare against)
 * Drawing Dispmanx layer with background and on top transparent OpenGL shapes (alpha_test.py --dispman)
 * Drawing Dispmanx layer wih the backgroun and on top transparent OpenGL shapes but seting the dispmanx element with the premult flag (alpha_test.py --dispman --premult)

As you can see the --dispman output differs a lot from the just OpenGL output (darker shapes) and with --premult it is pretty much the same as the OpenGL output.
It also generates some screenshots of the OpenGL buffer (had to change the pi3d screenshot function a bit).
From what I understand is that OpenGL blends everything with the background color (black in this case), so that semi transparent shapes get darker RGB values, and that the dispmanx ALPHA_PREMULT flag is compensating for that.

I don't have too much experience with this, nor is there too much information on the dispmanx behavior, so I'm not sure if there's a better way to fix this...
If you think this is correct I'm more than happy to make a pull request with it (the change in create_surface if quite trivial as you can see in the patched function).
Let me know what you think!

paddywwoof, 2016-08-22:
@swehner I've got round to looking at this - at last - sorry for the delay! 

Your fix seems to work really well. It's not something I've experienced previously but using pi3d for 2D overlays does seem to be something that a few people do so it makes sense to get it working as well as possible. If you get chance to do a pull request I will put it into the source.

Thanks again for contributing.

swehner, 2016-08-22:
Sure! No rush - thanks for taking a look at it.

The effect is really only noticeable when drawing things with a relatively low alpha values because the colors get darker - for anything where alpha is pretty much 1 or 0, like the Raspberry_Rain demo there's hardly any difference.
And I only noticed when switching from doing it purely in OpenGL to drawing the background in a separate dispmanx layer.

I've created a pull request for it here: #198 
Thanks!

closed 2016-08-23 by paddywwoof

swehner, 2016-08-23:
Hey - I was just wondering if you think it's worth it to make this configurable as a parameter 
If someone used differen

PR #198 (2016-08-23): Use PREMULT for the OpenGL dispman layer
Use the PREMULT flag for the dispmanx element for the OpenGL buffer

Fixes #197
commit 6884e921 (2016-08-23): Use PREMULT for the OpenGL dispman layer

Fixes #197