tipam/pi3d #201: 2d_flat.vs has incompatible z value

buxit, 2016-10-17:
the interpretation of the z axis in `2d_flat` shader is inconsistent with other shaders:
other shaders use:
`gl_Position = modelviewmatrix[1] * vec4(vertex,1.0);`

while `2d_flat` uses:
```
gl_Position = vec4(vertex, 1.0);
gl_Position.z += unif[0][2];
```
this is confusing and obscure and it makes it very hard to mix `2d_flat` shaded shapes with any other shapes (tested with an `is_3d=False` camera). maybe the code in `2d_flat.vs` could be changed to match the other shaders:
`gl_Position.z = (modelviewmatrix[1] * vec4(vertex,1.0)).z;`

if this is not possible, there might be a way to adapt the modelviewmatrix of the 2d mode so the z axis behaves consistently?

if none of this is possible, the meaning of `z` with `2d_flat` might just need more / better documentation. maybe there's an easy way to convert between the different z values?

paddywwoof, 2016-10-17:
@buxit yes the code in 2d_flat might be due an overhaul. It actually dates from BEFORE the idea of having a different Camera matrix for 2D rendering (using the ``is_3d`` argument) The shader uses the screen pixel locations rather than the uv coordinates interpolated from the vertex locations to map the textures.

The idea was that 2D icons, dials, mini-maps etc could be rendered in fixed screen positions as the first person camera moved around the scene. In fact it is nearly always better to use ``uv_flat`` shader along with a second Camera set ``is_3d=False``. I kept the 2d_flat in a) to not break anything made previously b) there might be some cases where the extra processing of swapping between two camera matrices impacts on frame rate (though I doubt it). I need to put more of this explanation into the FAQ!

What kind of effect are you looking for, and does ``uv_flat`` help?

buxit, 2016-10-18:
@paddywwoof i didn't think about performance when i decided to use the `2d_flat` shader for drawing a dynamic bar graph. i just liked how the shader places the origin in the top left corner, how all units are in pixels and how `set_2d_size()` makes the sprite grow towards bottom right. i'm using a copy of the `uv_flat` shader (modified to multiply color with first light) for a lot of other sprites.

now after what you say, i would opt to just deprecate the use of the `2d_flat` shader in the docs (possibly also adding that it can't easily be mixed with a camera set to `is_3d=False`).

paddywwoof, 2016-10-19:
@buxit  I will do that. One of the problems of mixing orthographic and perspective camera is the byproduct of calculating the z values that I explain here http://pi3d.github.io/html/FAQ.html#nearness-of-2d-relative-to-3d So it probably is possible to position 2d_flat planes using the messy formula, but not very  accessible.

PS did you try using the uv_light shader, that might do what your modified uv_flat shader does. Alternatively the set_material method can be used to alter the default rgb values of the Shape which are multiplied by the texture sampled values (see the Water.py demo)

closed 2016-10-19 by paddywwoof