tipam/pi3d #202: After running on pi, terminal is in a funny state.

stuaxo, 2016-10-19:
After I ctrl-C the terminal looks like this

```
Traceback (most recent call last):
                                    File "Minimal_2d.py", line 16, in <module>
                                                                                  while DISPLAY.loop_running():
        File "/home/pi/pi3d/pi3d/Display.py", line 203, in loop_running
                                                                           self._loop_end()  # Finish the previous loop.
                 File "/home/pi/pi3d/pi3d/Display.py", line 375, in _loop_end
                                                                                 self.swap_buffers()
                                                                                                      File "/home/pi/pi3d/pi3d/Display.py", line 401, in swap_buffers
                                                                self.opengl.swap_buffers()
                                                                                            File "/home/pi/pi3d/pi3d/util/DisplayOpenGL.py", line 263, in swap_buffers
                                                                 openegl.eglSwapBuffers(self.display, self.surface)
          KeyboardInterrupt
                           (lumen_client) pi@raspberrypi:~/pi3d_demos $ 

```

Pressing enter doesn't make a new line, but instead displays on the same line:

```
(lumen_client) pi@raspberrypi:~/pi3d_demos $ (lumen_client) pi@raspberrypi:~/pi3d_demos $
```

stuaxo, 2016-10-19:
Possibly the funny state is set when the program started, here is what the output of SpriteBalls.py for instance:
```
59.814759373098106
                  59.47803979131232
                                   59.478414266337836
```

paddywwoof, 2016-10-19:
@stuaxo not certain if this is your problem but I have had experienced similar breaking out when curses has been started. ``stty sane`` will normally fix it. http://stackoverflow.com/questions/11753909/clean-up-ncurses-mess-in-terminal-after-a-crash let me know if this is the problem (It's another thing that ought to go in the FAQ!)

stuaxo, 2016-10-19:
Running ```stty sane``` before did work. 

I don't know if there is a way for the program to reset the terminal state on exit ?

paddywwoof, 2016-10-19:
The 'standard' way I have seen this done is to wrap the code in a try/catch like

```
keys = pi3d.Keyboard()
mouse = pi3d.Mouse()
...
try:
  while display.loop_running():
...
    if keys.read() = 27:
      break

finally: #can also except KeyboardInterrup: for ctrl c specific things
  display.destroy()
  keys.close()
  mouse.stop() #why are all these different!!
```

stuaxo, 2016-10-19:
Any idea what is making making the terminal get into that state in the first place? 

Is it something to do with the ncurses input handling ?
 S++

rec, 2016-10-19:
ncurses sends various control characters to the terminal to put it into a
different state, which need to be undone by the sending of other control
characters at the end.

The try/finally block is standard, and indeed, basically the only way to do
it!

stuaxo, 2016-10-19:
Cool, is it just that some of the examples need this added ? S++

paddywwoof, 2016-10-19:
@stuaxo probably! I started off with the view that try/catch ought to be avoided unless absolutely necessary. However it seems to be regarded as 'pythonic' and quite ok!
All the demos should be escapable with the Esc character.

stuaxo, 2016-10-19:
Oh, I hadn't thought of pressing escape, as I was ssh'd in remotely - it does work.

The try/catch handler is probably pythonic as it's quite explicit about what is going on.  Another way is using atexit (though you have to remember to chain to the default handler as there is only one atexit handler.

rec, 2016-10-19:
Nah, atexit allows you to add as many functions as you like:
https://docs.python.org/2.7/library/atexit.html
https://docs.python.org/3/library/atexit.html

I think you're thinking of sys.exitfunc
https://docs.python.org/2/library/sys.html#sys.exitfunc (Python 2.x only, I
think).

stuaxo, 2016-10-19:
Oooh, cool :)

paddywwoof, 2016-10-19:
Hmm, maybe nice to add atexits to Display, Mouse and Keyboard. I will look into this some more. Thanks!

rec, 2016-10-19:
Not the right way!!!  Use a context manager.

Check this little commit I made:
https://github.com/tipam/pi3d/commit/c5aefda5c13f9bc3eab2b53200fbc3d1785820e6

NOTE NOTE NOTE:  entirely untested, I don't have an RP here..

stuaxo, 2016-10-19:
Very nice, *much* better than using atexit :)

paddywwoof, 2016-10-19:
Great. All tests run ok on pi with python2 and python3, and altering a few of the demos to use context manager structure allows programs to break but still close the curses nicely. I don't think Mouse or Display have any issues with breaking - the GPU memory seems to be reset by the process of shutting python or creating a new drawing context.

closed 2016-10-19 by paddywwoof