tipam/pi3d #203: Is there a way to tell which drivers pi3d is using?

stuaxo, 2016-10-20:
I've installed various libraries on my raspberry pi, the pi3d demos work, but I'd quite like to know which driver it is using.

In particular, because I have an app that uses gstreamer, ideally I'd like to integrate the two, but first I'd like to see what the state of play is when I run things right now.

paddywwoof, 2016-10-20:
I have to admit to not being entirely sure what the meaning of driver is here :$ on the RPi pi3d essentially just uses the python modules that you will have installed: numpy and PIL I don't know how much dependency these have on other installed libraries but I imagine they run their own instance of native executables.

Within pi3d there are three dynamically loaded libraries: libbc_host.so, libEGL.so and libGLESv2.so which are closed source Broadcom supplied. Different apps using OpenGLES on the pi all seem to use the same libraries and I'm not sure there's any conflict having multiple instance launched. You can run pi3d multiple times on the same pi and, apart from memory and processing requirements, I think they all work ok (have to set the relevant layer for the display surface).

If you run top from a different terminal while pi3d is and isn't running you can see very little extra processes start. There's python and VCHIQ-0 or VCHIQr-0 pop up from time to time.

Don't know if any of this is helpful or relevant!

stuaxo, 2016-10-20:
Ah, I think I was wondering what it is using for graphics - so I guess it will be doing libEGL stuff in this case.

- It's quite a while since I looked inside pi3d, + remembered that sometimes it can use pygame, was just trying to work out what is outputtng the gfx here.

stuaxo, 2016-10-20:
I was basically trying to work out if inside pi3d, there is enough stuff to pass to this API -

So that pi3d could provide an opengl es surface + "dispmanx window handle" so I could use it with this gstreamer API:  gst_gl_window_dispmanx_egl_set_window_handle

More info
https://bugzilla.gnome.org/show_bug.cgi?id=772608

paddywwoof, 2016-10-20:
Hard to say, If you look inside DisplayOpenGL around line 120 you will see various things that might match and would be available from the main app as
DISPLAY.opengl.surface
DISPLAY.opengl.nw_p
etc
I went round in circles trying to get a surface from the x server or pygame on the RPi that libEGL functions could turn into a drawing surface and failed!

stuaxo, 2016-10-20:
This is outside X,  .. it looks like I need something from "dispmanx" ,  but I only heard of dispmanx for the first time about a week ago.

I think it's basically a native raspberry pi 2d API.

I'll have a look, I guess DISPLAY.opengl.surface will be the egl surface in this case ?

swehner, 2016-10-20:
Hi,

If you want some more info - here's good documentation on dispmanx and all things graphics related on the pi:
https://jannewmarch.gitbooks.io/programming-audiovideo-on-the-raspberry-pi-gpu/content/index.html

In our project we ended up combining pi3d together with some other dispmanx layers to draw the background.
https://github.com/swehner/foos/blob/master/foos/ui/bg.py

Another thing I've tried is playing video using the ILClient on a GLtexture defined in pi3d (similiar to the hello_videocube demo, but from Pi3D). It's pieces I've hacked together from the demo programs, but it works decently.
https://github.com/swehner/foos/blob/opengl_replay/foos/ui/opengl_replay.py
https://github.com/swehner/foos/blob/opengl_replay/egl_replay/video_helper.c

Hope that helps you! At least the Jan Newmarch book is a nice read :)

paddywwoof, 2016-10-20:
swehner, that's all really good, useful info. When I set up the video texture I looked at the Broadcom hello_videocube/teapot demo but decided it would be nightmare to pythonise - how sensible to leave it as c and import the so.

stuaxo, 2016-10-21:
That book looks really interesting, unfortunately most links I try seem to 404.

I'll certainly have a look, possibly through archive.org if nowhere else.

stuaxo, 2016-10-24:
Turns out it is going to be published soon, so will buy when it comes out, in the meantime the author has been kind enough to provide the info.

paddywwoof, 2016-10-24:
@stuaxo, will close this for tidiness now. Resulted in some good info!

closed 2016-10-24 by paddywwoof