tipam/pi3d #204: Pass in numpy array as texture ?

stuaxo, 2016-10-24:
This is probably a noob question, but is it straightforward to pass in a texture from a numpy array ?

Specifically I have a numpy array with some fft data, and hoping to make some audio reactive things, where I pass that (or just plain wave data) in.

paddywwoof, 2016-10-24:
Hi, yes it's easy to pass a numpy array to a texture, just:
``tex = pi3d.Texture(grfx)``
where grfx is a numpy array with shape = (H,W,N) where N is 3 or 4 (or possibly 1 though I've not tried that!) and dtype = np.uint8
HOWEVER from what you describe this might not be the best way as, even using https://github.com/tipam/pi3d/blob/master/pi3d/Texture.py#L274 takes a bit of time as an image is a reasonable sized array. Instead you could create a ``pi3d.Lines`` instance with a much smaller numpy array. You would then refresh the locations of the vertices with https://github.com/tipam/pi3d/blob/master/pi3d/Shape.py#L454 which I think would be quicker (if not too much else going on you can get 20 to 30fps even on the B) I don't know what fft you have but you could generate a 3D 'strip' where the frequency bands went at right angles to time. Psychedelic!

closed 2016-12-23 by paddywwoof

stuaxo, 2016-12-23:
Cheers, for the info :)