tipam/pi3d #205: Integrate with gstreamer (reprise) ?

stuaxo, 2016-10-31:
As an output of the conversation here
https://github.com/tipam/pi3d/issues/203

I've managed to get gstreamer to output to a dispmanx window I created, using set_window_handle

https://gist.github.com/stuaxo/3009b761cba1085ac5730c7485506ee4

Now I have a this, is it possible to use this with pi3d, to do stuff like apply shaders, or even use it a video texture somehow ?

paddywwoof, 2016-10-31:
You might be able to disable the part of pi3d/util/DisplayOpenGL.py that does the same native window creation and pass the handle to pi3d from your glstreamer code. However I imagine that there might be issues updating the same surface by two agents. If you want to use the video as a Texture then I think it would be easier to use the method in swehner's foos. You can create a second (and more) display surface for pi3d to draw onto that will be in front of other surfaces.

stuaxo, 2016-10-31:
I'll have a look into these, cheers :)

closed 2016-10-31 by stuaxo