tipam/pi3d #206: hi target name?

darknesspider, 2016-12-21:
How to find the name of the object in the camera's target (pi3D 3d world )
example:https://www.youtube.com/watch?v=0u91pcNXtcI&feature=autoshare
sun,world,astroid123...

paddywwoof, 2016-12-21:
As I commented on youtube not entirely trivial, however familiarity with vectors and numpy will help. 

One approach would be to look at the size of the angle between the direction to each possible target and the direction the camera is pointing. You would then choose objects where that angle was smaller than a threshold then choose the nearest if more than one. Say in pi3d_demos.ForestWalk.py you had kept a dict of different objects as obj_dict, you could add a print() when you press the 'h' key:
```
...
    elif k == ord('h'):
      nearest = None
      target_name = None
      for o_name in obj_dict:
        o = obj_dict[o_name]
        v1 = o.unif[0:3] - CAMERA.eye # numpy vector from camera to object
        dist = (v1 ** 2).sum() ** 0.5
        v1 /= dist # divide by length
        v2 = CAMERA.get_direction() # already length = 1
        cos_angle = v1.dot(v2) # 1 straigt at it, 0 at right angles, -1 opposite direction
        if cos_angle >= THRESHOLD:
          if nearest is None or dist < nearest:
            nearest = dist
            target_name = o_name
      print('nothing' if target_name is None else target_name)

```
However this only works for smallish objects. If you want to see if the cross-hairs are hitting the Earth on the meteorize game then you would either have to include a size factor into the obj_dict above instead of using an overall threshold i.e. 
```
o = obj_dict[o_name][0]
sz = obj_dict[o_name][1]
if cos_angle > (1 - (sz / dist) ** 2) ** 0.5
```
Or look at each face of the object and see if the vector intercepts each triangle - but this would be very slow. See the Pong demo and the ElevationMap.clashTest() method.

Finally a fairly efficient but technical method would be a variation of the Clashtest system using glScissor to draw a single pixel of each object to an offscreen texture with different RGB values for each then checking what if any had been drawn.

closed 2016-12-21 by paddywwoof

darknesspider, 2016-12-22:
thankss :)
i try dogfight but o.unif[0:3] crash

paddywwoof, 2016-12-22:
Hi, maybe I should have included the prior construction of obj_dict! In example one it was
```
obj_dict = {'monument':monument, 'trees1':mytrees1, 'trees2':mytrees2, 'trees3':mytrees3}
THRESHOLD = 0.95
```
which obviously has to happen after the Shape instances have been created. For the second idea it would need to be
```
obj_dict = {'monument':[monument, 4], 
        'trees1':[mytrees1,15], 
        'trees2':[mytrees2,20], 
        'trees3':[mytrees3, 20]}
```
With this example there would be issues due to being able to get 'inside' the tree objects, but you get the idea!

PS the dogfight algorithm for hitting a target is probably much clunkier than this and I might bring it up to date!

darknesspider, 2016-12-23:
thank you very much :)
and i need help.
how to create bullet and shoot target?

paddywwoof, 2016-12-23:
Hi, the bullets in Counter Strike just look like streaks of light with some smoke so the best way is probably something like the bullets in DogFight. This uses two Plane objects merged at 90 degrees to each other with a series of images that can be mapped onto them in sequence (there are two sets of guns on the Aeroplane which is why there are four Planes). How you animate the target being shot depends on the effect you want. In the meteorize game I use a part transparent texture and scale the meteors up so they look to be exploding.

darknesspider, 2016-12-27:
Hello again
how to create laser point for gun?
thank you...

paddywwoof, 2016-12-27:
Hi, that's also non-trivial. The easiest way is probably to make a custom fragment shader that adds a red component to pixels in the centre of the screen with a radius dependent on the depth. The disadvantage is that there would be a bit more processing time for all pixels. Probably not too severe. I will post some code to show you how to do it. 

PS Although your questions aren't really "issues" they might be useful references for others a and I will add them to the FAQ

paddywwoof, 2016-12-27:
@darknesspider, OK this is the kind of thing you can do. For example in ForestWalk.py:
1. copy the shaders you want to get lasers on to the pi3d_demos/shaders directory and rename i.e. uv_bump_laserdot.vs, uv_bump_laserdot.fs
2. edit the fragment shaders i.e. uv_bump_lasterdot.fs and add the two lines
```
#include std_bump.inc
  float radial = distance(vec2(gl_FragCoord), unif[15].xy); // 1 --- distance from centre of screen
  texc.rgb += vec3(5.0, -2.0, -2.0) / (dist * radial); // 2 --- increase R, decrease GB reduce size in distance
  gl_FragColor =  (1.0 - ffact) * texc + ffact * vec4(unif[4], unif[5][1]); // ------ combine using factors
```
3. change ForestWalk.py to use the new shaders
```
# load shader
shader = pi3d.Shader("shaders/uv_bump_laserdot")
...
```
4. change ForestWalk.py to add the x, y location of the screen midpoint to each Shape that uses the new shaders
```
...
mymap.set_fog(*FOG)
mymap.unif[45:47] = [DISPLAY.width / 2.0, DISPLAY.height / 2.0] #or use mymap.set_2d_size(DISPLAY.width / 2.0, DISPLAY.height / 2.0)
```

It also occurs to me that you could take an alternative route if you are using the obj_dict idea from above. In that case you could simply draw the laser dot in the middle of the screen (i.e. using a Camera(is_3d=False..)) and scale the dot according to distance to the target. The disadvantage is that the dot wouldn't scale correctly on very large objects such as the terrain.

darknesspider, 2016-12-28:
thank you very much. very nice
But it does not show up on the tree or other objects.
and how can I learn the name of the object on it
I'm sorry to ask so many questions

paddywwoof, 2016-12-28:
The trees use a different shader uv_flat and the monument uses uv_reflect You need to make modified shaders for everything you want to show a laser dot (and set unif[45:47] in each) 

There is no way (easily) to get info back from the GPU (apart from my last suggestion for using off screen textures) so to determine where the dot is you need to do something like the obj_dict checking I suggested foot your first question. However this won't give you useful info for large merged objects like the trees.

darknesspider, 2016-12-28:
thank you very much