tipam/pi3d #210: Wrong event handling

int-0, 2017-04-12:
Hi,
I've written a simple loop to test EventHandler(). I check a keyboard handler and a mouse handler. Everything looks fine, but when I move the mouse I get the following error:
```
Traceback (most recent call last):
  File "../src/__init__.py", line 62, in update
    self.inputs.do_input_events()
  File "build/bdist.linux-x86_64/egg/pi3d/event/Event.py", line 103, in do_input_events
    if self.handler.event(event) and self.unhandledHandler:
  File "build/bdist.linux-x86_64/egg/pi3d/event/EventHandler.py", line 94, in event
    self.rely[event.stream.deviceIndex] += event.eventValue
IndexError: list index out of range
```
After some investigation, I get the wrong deviceIndex value, this value is 10 and self.rely is a 4-element list. And also I print the event information that causes this issue. The event is:
`DEBUG:pi3d.event.EventHandler:Input event keyboard[10], 1 -- 1492019728.066793: 0x2(0x1) = 0x2`

As you can see, this event is "keyboard" but is not true, this event is emited when I move the mouse.
I can try to fix by myself but is very hard for me without more documentation, sorry...

I have more than one mouse (a Wacom tablet with 3 pointing devices). I can send you a dump of my _/proc/bus/input/devices_ file if you want.

Could you help me? can I help you?

Thank you!

paddywwoof, 2017-04-12:
Sounds like this does need sorting out. I'm away at the moment and won't get back till Saturday night but will have a look when I get home.

I have found that I run into variable behaviour when running pi3d on different laptops which allocate different pointing devices. I don't know if it's possible to make this automatically fixable but it is necessary to have a clearer explanation of what's going on so people can tweak the setup more easily. (The event system was devised by Rurwin and is rather sophisticated (i.e. I've never really got the bottom of how it all works!))

paddywwoof, 2017-04-16:
@int-0, Hmm now I look at it again, the event system is certainly complicated! Maybe if you give me a copy of your simple loop I will try and reproduce the error. And, yes, /proc/bus/input/devices would be useful too as that looks to be where find_devices() generates deviceIndex

int-0, 2017-04-18:
Hi,
First of all... thank you!

Ok, I get a following error when I move the mouse (not the tablet!):
```Traceback (most recent call last):
  File "test_inputs.py", line 20, in <module>
    inputs.do_input_events()
  File "build/bdist.linux-x86_64/egg/pi3d/event/Event.py", line 103, in do_input_events
    if self.handler.event(event) and self.unhandledHandler:
  File "build/bdist.linux-x86_64/egg/pi3d/event/EventHandler.py", line 88, in event
    self.relx[event.stream.deviceIndex] += event.eventValue
IndexError: list index out of range
```
And the code is:
```python
#!/usr/bin/env python
#

import traceback
import pi3d

def keyboard_handler(sourceType, sourceIndex, key, value):
    print sourceType, sourceIndex, key, value

def mouse_handler(sourceType, sourceIndex, x, y, v, h):
    print sourceType, sourceIndex, x, y, v, h

display = pi3d.Display.create(w=1024, h=768, frames_per_second=20.0)
inputs = pi3d.event.Event.InputEvents(keyboard_handler, mouse_handler)

cycles = 100
while (display.loop_running()) and (cycles > 0):
    try:
        inputs.do_input_events()
    except Exception, e:
        traceback.print_exc()
    finally:
        cycles -= 1
```

And also, my _/proc/bus/input/devices_ file:
```I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input3
U: Uniq=
H: Handlers=kbd event0 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input4
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A03:00/device:03/LNXVIDEO:01/input/input5
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0
[... 237 lines cut ...]

I: Bus=0003 Vendor=056a Product=0028 Version=0110
N: Name="Wacom Intuos5 touch L Pad"
P: Phys=usb-0000:00:12.2-1.3/input0
S: Sysfs=/devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.3/1-1.3:1.0/0003:056A:0028.0015/input/input55
U: Uniq=
H: Handlers=event25 js0 
B: PROP=0
B: EV=b
B: KEY=800 1ff 0 0 0 0
B: ABS=10000000103

I: Bus=0003 Vendor=056a Product=0028 Version=0110
N: Name="Wacom Intuos5 touch L Finger"
P: Phys=usb-0000:00:12.2-1.3/input1
S: Sysfs=/devices/pci0000:00/0000:00:12.2/usb1/1-1/1-1.3/1-1.3:1.1/0003:056A:0028.0016/input/input57
U: Uniq=
H: Handlers=mouse3 event26 
B: PROP=1
B: EV=b
B: KEY=e520 0 0 0 0 0
B: ABS=263800000000003

```
You can send me code snippets or anything you need to run in my system... thank you again!

paddywwoof, 2017-04-18:
Hi, thanks for that. The event.FindDevices.find_devices() function looks to go through the input/devices file and generate a list depending on 'kbd' 'mouse' or 'js' occurring in the line beginning 'H' which means that on your system it should get up to a total of 11 keyboards, 4 mice and 1 joystick. A lot of mice seem to also call themselves keyboards - presumably they have some keyboard functionality built into them.

Anyway, I managed to reproduce your error after a fashion by plugging more and more mice and graphics tablets into my laptop. The problems seemed to be solved by increasing the list sizes in EventHandler.py

I have pushed a modified version to github develop branch. You can try it out by cloning from github, checkout develop then adding

```
import sys
sys.path.insert(1, '/home/int0/pi3d') #or whereever
import pi3d
```

int-0, 2017-04-18:
yeah, you rocks!

Now it works... anyway I'm reading some code in [python-evdev](https://github.com/gvalkov/python-evdev) and it seems that this _magic_ number can be detected. But for me (at the moment) is out of scope... if you are agree this issue could be closed... :)

int-0, 2017-04-19:
Hahaha... mouse events are reported as "keyboard" instead of "mouse"... but is fine for me,

Thank you again! ;)

paddywwoof, 2017-04-19:
Yes, I've not tried to figure out how the pi3d.event system differentiates kbd mouse js where they're all on the same device. As you say it probably doesn't matter! As you're happy I will close the issue. (and move the master branch to v2.19)

closed 2017-04-19 by paddywwoof