tipam/pi3d #214: GLES libraries have changed their names in Raspbian Stretch

gpjt, 2017-09-03:
I've just put in [a pull request](https://github.com/tipam/pi3d/pull/213) with a proposed fix for this, but will log this issue too just in case anyone is hitting the same problem and wondering what's going on.

In the most recent version of Raspbian, code name "Stretch", the non-open-source Broadcom OpenGLES 2.0 library was renamed from "libGLESv2.so" to "libbrcmGLESv2.so" and likewise "libEGL.so" was renamed to "libbrcmEGL.so", [to avoid name clashes with the MESA library](https://github.com/RPi-Distro/firmware/issues/3).  

This means that pi3d doesn't initialize properly, so for example you get an error like this when running the Earth demo:

```
(pi3d) giles@percy:~/pi3d_demos$ python Earth.py 
Traceback (most recent call last):
  File "Earth.py", line 11, in <module>
    import pi3d
  File "/home/giles/.virtualenvs/pi3d/lib/python3.6/site-packages/pi3d/__init__.py", line 10, in <module>
    from pi3d.constants import *
  File "/home/giles/.virtualenvs/pi3d/lib/python3.6/site-packages/pi3d/constants/__init__.py", line 170, in <module>
    PLATFORM, bcm, openegl, opengles = _detect_platform_and_load_libraries()
  File "/home/giles/.virtualenvs/pi3d/lib/python3.6/site-packages/pi3d/constants/__init__.py", line 165, in _detect_platform_and_load_libraries
    set_gles_function_args(opengles) #function defined in constants/gl.py
  File "/home/giles/.virtualenvs/pi3d/lib/python3.6/site-packages/pi3d/constants/gl.py", line 381, in set_gles_function_args
    gles.glActiveTexture.argtypes = [c_uint] #GLenum
AttributeError: 'NoneType' object has no attribute 'glActiveTexture'
```

The fix is to rename the libraries in `pi3d/constants/__init__.py`.

paddywwoof, 2017-09-03:
develop version now fixed. Will close this out when I've bumped version to 2.21

Thanks again.

paddywwoof, 2017-09-13:
pi3d updated to v2.21 with this pull incorporated

closed 2017-09-13 by paddywwoof