tipam/pi3d #215: EGL_NO_SURFACE set to 0, but openegl.eglCreateWindowSurface returns None on error

gpjt, 2017-09-13:
This is because in egl.h, there is:

    egl.eglCreateWindowSurface.restype = c_void_p

Probably best to override EGL_NO_SURFACE to be None, though I'm not sure whether that would be best done in `constants/egl.py` or in `constants/__init__.py` (suspect the latter, but...)

gpjt, 2017-09-13:
Just updated issue title: looks like the same applies to EGL_NO_DISPLAY / `openegl.eglGetDisplay`

paddywwoof, 2017-09-13:
giles, thanks. Presumably this is to get the assertions to catch the case where there is no display or window surface. Does it just work with None or do you have to treat it with some ctypes such as .value or c_void_p()? I can't get eglGetDisplay() to return anything but an integer (lots of segmentation faults trying!)

paddywwoof, 2017-09-13:
Just noticed that EGL_NO_SURFACE and EGL_NO_CONTEXT get passed to eglCreateContext() and eglMakeCurrent() so might need to do something more complicated
`assert self.surface != EGL_NO_SURFACE and self.surface is not None`
 ?

gpjt, 2017-09-14:
Right, I guess this one must be a tricky one to repro if your eglGetDisplay is behaving itself!   I posted this after spending a while trying to debug an issue on my desktop (Linux 64-bit), where the underlying cause turns out to be that the `eglGetDisplay` always returns `EGL_NO_DISPLAY` due to driver issues.

So, yes, the problem this is meant to work around is that in `DisplayOpenGL` we have this code:

```
  def create_display(self, x=0, y=0, w=0, h=0, depth=24, samples=4, layer=0, display_config=DISPLAY_CONFIG_DEFAULT):
    self.display = openegl.eglGetDisplay(EGL_DEFAULT_DISPLAY)
    assert self.display != EGL_NO_DISPLAY

```

The assertion does not fail when `EGL_NO_DISPLAY` is `0`, even though `eglGetDisplay` is returning NULL, because the NULL is being converted to None (and of course `None != 0`).

I think that setting both `EGL_NO_SURFACE` and `EGL_NO_CONTEXT` to `None` should work fine -- both `eglCreateContext` and `eglMakeCurrent` have `c_void_p` as the relevant `argtypes`.

paddywwoof, 2017-09-14:
Hi, I've just tried that and it crashes out at the eglCreateContext() call. For the moment I have changed the assert lines as per comment above (with additional condition checking for None)

closed 2017-09-14 by paddywwoof

gpjt, 2017-09-14:
Thanks!  How odd, though.  I'd have sworn it would work, but I'm obviously wrong.

I'll do a bit more poking around and will post back here or submit a PR if I find something that works in both cases,