tipam/pi3d #216: Shader/Loadable Thread Dependence

parallellogic-, 2018-03-11:
When I run `self.shader_3d = self.pi3d.Shader("uv_light")` I receive an assertion error with no descriptive text.
After some digging, it appears that the first call in `pi3d.Shader.__init__()` is `Loadable.is_display_thread()`.  This method in turn inspects whether the thread where the pi3d library was loaded from is equal to the thread that the Shader is currently being instantiated in.
This caused me an issue (AssertionError) because I loaded pi3d in the __init__() method of a thread (a method call which lives in Python's Main thread space) but then created a shader in the body of the thread (the run() thread space of the child thread).
I have since moved the pi3d load to the beginning of the run() method so that pi3d is loaded in the same thread (child thread) where it is being used (child thread).  However, I'm uncertain: is this the intended behaviour?  Am I supposed to be forbidden from loading pi3d in one thread space and then using it in another?

parallellogic-, 2018-03-11:
By "loaded pi3d", I mean:
`import sys
sys.path.insert(1, '/home/pi/pi3d')
import pi3d`

paddywwoof, 2018-03-11:
Hi @parallellogic- yes it's an annoying thing to do with the implementation of the libELG or libGLES2 shared libraries. @rec (and others) banged their heads against it as described here https://github.com/tipam/pi3d/issues/13

I can't remember which things have to happen in the main thread and which things don't but I think the Display loading, Texture loading, Shader loading and probably Shape.draw() all do.

When I've had issues with other modules also needing to be in the main thread (i.e. Flask) I've got it to run ok using multiprocessing which is designed to have a close syntax to threading so not so bad. But there is an issue passing info between processes which is a bit trickier. I put some notes on that here
http://pi3d.github.io/html/FAQ.html#display-size-greater-than-2048

I will leave this issue open as I think there might need to be a bit more info with the assertion failure.

parallellogic-, 2018-03-11:
Dang you're fast; I appreciate that.  Ok, just double-checking.  Thanks for the links, I'll keep that in mind if I encounter any further issues.

> might need to be a bit more info with the assertion failure

Detailed stack trace:
`  File "/home/pi/utils/Resources.py", line 166, in __create3Dgraphics`
`    self.shader_3d = self.pi3d.Shader("uv_light")`
`  File "/usr/local/lib/python3.5/dist-packages/pi3d/Shader.py", line 80, in __init__`
`    assert Loadable.is_display_thread()`
`AssertionError`

paddywwoof, 2018-03-23:
Hopefully will provide a bit more info in future https://github.com/tipam/pi3d/commit/82dd6384ae5251179ed63191afb8da1c018bfc32

closed 2018-03-23 by paddywwoof