tipam/pi3d #217: ZXY Euler Angle Rotation Sequence in Shape

scottalmond, 2018-03-19:
I'm looking to define a coordinate system that is offset from the origin in the y-dimension, then rotated about the y-axis, and then translated along the local y-z plane.  In that coordinate system I'm looking to rotate a shape about the local x-axis.  However, my attempt to do has resulted in unexpected behavior when rotating and I'm wondering if it's a byproduct of the fact Pi3D uses an ZXY Euler angle sequence of rotations in Shape, assuming I am reading `if self.rozflg:` etc from the following correctly:
https://pi3d.github.io/html/_modules/pi3d/Shape.html

I'm doing the vector transforms and matrix math using the transforms3d library and only at the end converting back to Euler angles to plug into Pi3D.Shape.  I suspect there's something getting lost in translation between my matrix math and the back calculation of the ZXY Euler angles.  Is there any other way to rotate an object other than through the roz/rox/roy transforms in Pi3D.Shape?  I'm wondering what would happen if I were to supply the transformation matrix directly.

For a concrete example, I have a player cargo pod with a laser on it (the laser is translated from the origin in the y-dimension).  The laser base is supposed to rotate (local y-axis) in one dimension and then the laser gun (whose pivot is translated in the y-z plane) is supposed to gimbal (local x-axis) in another.  However, when I attempt to perform these transforms, I note the laser gun rotates away from the base.
https://www.youtube.com/watch?v=Ov9-FUBi-ys

The code I'm using is below, namely the `__getOrientation` method
https://github.com/scottalmond/EscapeRoom/blob/master/src/chapters/wall/hyperspace_helper/PlayerPod.py

Any help is appreciated, thank you

paddywwoof, 2018-03-19:
@scottalmond just back now and will look at this. I went round in circles when first involved with pi3d as some of the matrix stuff that seemed to work was wrong (different from the way I understood it and it's generally described in maths books). I came to the conclusion that the issue was related to the fact that the OpenGL convention is different from standard set of left hand axes with x going l to r, y going b to t and z going *into* the screen. So for the transforms3d convention (standard maths, after a very quick look at the docs) the mapping from pi3d would be x->X, y->Z, z->Y (or probably equivalent z->-Z). But... I have to say I messed around with transforming frames of reference in the early days and it never seemed to behave the way I expected.

The expanded matrix multiplication in Shape is effectively
T2(S(Ry(Rx(Rz(T1)))))

The child Shape system (i.e. gun and turret on Tank demo) works well for keeping the articulated parts attached (and will be fast to calculate) but it still has a problem of mapping back from an absolute frame of ref to the local rotated frame. Using quaternions is the standard way of fixing this (to some extent) and it might be worth adding a bit of functionality to that end. (There is some functionality in the Camera class I seem to remember so that might be extendable.)

I will get back to you when I've looked at your code. Will I be able to clone your repo and 'just' run it?

paddywwoof, 2018-03-19:
Couldn't see an easy way of testing just the bit of your code giving the issue. Here's an alternative way of approaching it. There are some messy bits to get direction vectors of child shapes which might be better wrapped in methods. Also if you need to use transforms3d for other things then you might need to stick to the method you've developed.

```
import pi3d
import numpy as np

RANGE = 100.0
MODEL_PATH = '/home/patrick/raspberry_pi/EscapeRoom/src/chapters/wall/assets/hyperspace/{}'
display = pi3d.Display.create(background=(0.0, 0.0, 0.0, 0.0))
shader = pi3d.Shader('uv_light')
laser_shader = pi3d.Shader('mat_flat')

# pod and laser
pod = pi3d.Model(file_string=MODEL_PATH.format('pod_2.obj'), z=20.0)
laser_base = pi3d.Model(file_string=MODEL_PATH.format('laser_base_2.obj'), y=3.15)
laser_gun = pi3d.Model(file_string=MODEL_PATH.format('laser_gun_2.obj'), y=0.4, z=-0.4)
laser_tip = pi3d.Cylinder(radius=0.05, height=5.0, rx=90.0, y=0.28)
laser_tip.set_material((1.0, 0.0, 0.0))
laser_tip.set_alpha(0.0)
laser_gun.add_child(laser_tip)
laser_base.add_child(laser_gun)
pod.add_child(laser_base)
pod.set_shader(shader) # all use uv_light shader
laser_tip.set_shader(laser_shader)

# asteroid
asteroid = pi3d.Model(file_string=MODEL_PATH.format('asteroid_large_1.obj'), z=50, y=10, x=20)
asteroid.set_shader(shader)

keys = pi3d.Keyboard()
mouse = pi3d.Mouse(restrict=False)
mouse.start()

blast_dist = 0.0

while display.loop_running():
  pod.draw()
  asteroid.draw()
  asteroid.rotateIncX(1.5)
  asteroid.rotateIncZ(2.0)
  mx, my = mouse.position()
  laser_base.rotateToY(mx)
  laser_gun.rotateToX(max(min(20, -my), -85))
  if blast_dist > 0.0:
    laser_tip.set_alpha(0.8)
    laser_tip.positionZ(2.5 + blast_dist)
    blast_dist += 5.0
    if blast_dist > RANGE:
      blast_dist = 0.0
      laser_tip.set_alpha(0.0)
  k = keys.read()
  if k >-1:
    if k == ord('a'):
      pod.translateX(-0.1)
    elif k == ord('d'):
      pod.translateX(0.1)
    elif k == ord('w'):
      pod.translateY(0.1)
    elif k == ord('s'):
      pod.translateY(-0.1)
    elif k == ord('z'):
      pod.translateZ(0.1)
    elif k == ord('x'):
      pod.translateZ(-0.1)
    elif k == ord('j'):
      pod.rotateIncY(2)
    elif k == ord('l'):
      pod.rotateIncY(-2)
    elif k == ord('i'):
      pod.rotateIncX(2)
    elif k == ord('k'):
      pod.rotateIncX(-2)
    elif k == ord(' '): # shoot with space
      blast_dist = 0.5 # positive value 'sets it off'
      ''' hit test check direction of pulse '''
      # translation needs 4x4 matrix and x,y,z,w with w set to 1.0
      tip_pt = np.dot(laser_tip.MRaw.T, [0.0, 1.0, 0.0, 1.0])[:3] # cylinder originally vertical now rotated in line with barrel
      but_pt = np.dot(laser_tip.MRaw.T, [0.0, 0.0, 0.0, 1.0])[:3] # needs to be transposed for use outside OpenGL
      aim_vec = tip_pt - but_pt # because translated need to do this
      dir_vec = asteroid.unif[0:3] - but_pt # vector to asteroid
      # dir_vec = asteroid.xyz - but_pt # can do this after pi3d v2.21
      len_dir = np.dot(dir_vec, dir_vec) ** 0.5 # std length calc
      if len_dir > 0: # just in case laser and asteroid are in same place!
        dir_vec /= len_dir # normalise
        c_prod = np.cross(dir_vec, aim_vec) # cross product of two vectors
        sin_ang = np.dot(c_prod, c_prod) ** 0.5 # magnitude of cross product is sin angle between as both unit vecs
        print(sin_ang, c_prod)
      ''' alternatively the laser pulse wouldn't be a child object but would be 'top level'
      the absolute direction vector of the gun would be got in the same way as above and 
      this would be used to increment the position of the pulse. The location of the
      pulse would then have to be checked each frame against the position of possible
      targets. In many ways this would be better but it would either need a spherical
      pulse or the rotation matrix would have to be extracted from the gun's combined
      transformation matrix.
      '''
    elif k==27:
      keys.close()
      display.destroy()
      break
```

scottalmond, 2018-03-20:
>The child Shape system (i.e. gun and turret on Tank demo) works well for keeping the articulated parts attached
Oh!  There's an "add_child" method, yeah, that makes things a lot easier.

>the mapping from pi3d would be x->X, y->Z, z->Y
Heh, I was noting something where +Z in Blender (.obj file) corresponds to +Y in Pi3D but wasn't looking too closely, I presume that's related.

>Will I be able to clone your repo and 'just' run it?  ... Couldn't see an easy way of testing just the bit of your code giving the issue.
Sorry about that, wasn't thinking ahead.  For reference, I have a section on the main page under "Dependencies/Software" that lists the supporting libraries needed to run the whole program.  The last step of this development effort will be fresh install on a clean RPi, so I'm trying to keep that somewhat up-to-date
https://github.com/scottalmond/EscapeRoom
For testing purposes, I've copied your code here:
https://github.com/scottalmond/EscapeRoom/blob/master/testing/test20_3d_rotation.py

>Also if you need to use transforms3d for other things then you might need to stick to the system you've developed.
Heh, I'm not tied to transforms3d.  I was just looking for a tested library since the first thing I mess up in matrix math is the transform matrices, dropping a sign on a sin, etc.

>Here's an alternative way of approaching it.
Haha, I only got those asteroids uploaded last night, you got to code against them before I could!

I greatly appreciate your example code, this directly addresses the issue I'm having, thank you.

@paddywwoof, since we're on the topic, do you have any suggestions for how to structure a game like the following?
https://www.youtube.com/watch?v=Er0AzrrjrJI&feature=youtu.be&t=14m47s
The main thing I'm wondering is if you have any suggestions on how to structure the relationships "add_child" between all objects in a playing field.

https://camo.githubusercontent.com/c2ee49899fcc2bef269fa32c8ee42efb9291f064/68747470733a2f2f692e696d6775722e636f6d2f6f486e614270462e6a7067

Currently I'm headed toward a split architecture (I'm defining a strand as computations relative to the global reference frame).  On one strand the player pod is free to translate in a 2D plane parallel with the screen, but otherwise remains fixed in space, and the camera is fixed in space pointed forward near the pod.  On the other strand the sequence of rings are all connected together and moved/rotated towards the player to simulate motion - like sucking up a strand of spaghetti (rings moving towards the player look like the player moving through the rings).

Conversely, another architecture is a single strand where the rings are the parent, the pod is a child that physically moves through space, and the camera follows the pod.  As the player moves through the rings, the rings behind are removed from the render sequence, and new ones are added ahead.  This feels like the more appropriate path, though it feels more involved since it involves working out how the pod and camera are oriented in space, rather than just keeping them fixed and having the rings move around them.

paddywwoof, 2018-03-20:
Hi, lots of ideas and questions. First, I've posted at the bottom the code for the different approach to making the laser_tip for checking hits using an animated particle. I've shown how you can extract the gun's rotation using the functions in Camera and use those to rotate the 'beam'. Where I use Shape.unif[] etc there are wrappers in pi3d such as Shape.xyz and Shape.X() that would be nicer in code that was designed for people to read!

You will see that it's not trivial to extract then compare translations from different levels in a parent-child hierarchy so I would err on the side of using this structure only where Shapes are logically components of a larger unit. (wheels on vehicle, legs on animal etc). 

I'm not sure which way I would go for the animated rings. Superficially they are the same but I suspect the collision detection would be easier if you restricted everything to the xy plane (apart from laser shooting) and made the rings move towards you. However you then would need to gradually rotate the rings so x and y rotations were zero as they passed the camera, but that might not be too tricky to do. To get 'nice' rotation about local z axis you will need to do as you say: make each ring a child of an 'empty' (Triangle with alpha=0). You can draw() the same object lots of times per frame, so you only need to have one ring and draw it at a series of six or so locations and rotations which you update each frame at the point z value goes below a threshold check distance of pod from centre then move ring to end of list and recalc location and rotation from last ring in list. And the same kind of thing with asteroids. More efficient to draw solid objects near the camera first to save GPU redrawing pixels.

pi3d (well python on RPi) won't be able to do calculations for a large number of objects. 20 or 30 will start to have a serious impact on performance.

The Shape.add_child() method is a simple wrapper for `self.children.append(child)` where children is just a list. So removing a child would be just `eg_empty.children.remove(eg_ring)`

```
...
HIT_DIST = 5.0
...
camera = pi3d.Camera()
...
# pod, base and gun
pod = pi3d.Model(file_string=MODEL_PATH.format('pod_2.obj'), z=20.0)
laser_base = pi3d.Model(file_string=MODEL_PATH.format('laser_base_2.obj'), y=3.15)
laser_gun = pi3d.Model(file_string=MODEL_PATH.format('laser_gun_2.obj'), y=0.4, z=-0.4)
laser_base.add_child(laser_gun)
pod.add_child(laser_base)
pod.set_shader(shader) # all use uv_light shader

# laser beam itself not a child
laser_tip = pi3d.Cuboid(w=0.05, h=0.05, d=5.0)
laser_tip.set_material((1.0, 0.0, 0.0))
laser_tip.set_alpha(0.8)
laser_tip.set_shader(laser_shader)
...
while display.loop_running():
  pod.draw()
  asteroid.draw()
  asteroid.rotateIncX(1.5)
  asteroid.rotateIncZ(2.0)
  mx, my = mouse.position()
  laser_base.rotateToY(mx)
  laser_gun.rotateToX(max(min(20, -my), -85))
  if blast_dist > 0.0:
    laser_tip.draw()
    laser_tip.position(*(tip_pt + aim_vec * blast_dist))
    blast_dist += 5.0
    if blast_dist > RANGE:
      blast_dist = 0.0
      asteroid.set_material((0.5, 0.5, 0.5))
    if (abs(laser_tip.unif[0] - asteroid.unif[0]) < HIT_DIST and # cheaper calc than euclidean dist
        abs(laser_tip.unif[1] - asteroid.unif[1]) < HIT_DIST and
        abs(laser_tip.unif[2] - asteroid.unif[2]) < HIT_DIST):
      euclidean = sum((laser_tip.unif[i] - asteroid.unif[i]) ** 2 for i in range(3)) ** 0.5
      print(euclidean)
      asteroid.set_material((1.0, 1.0, 0.0))

  k = keys.read()
  if k >-1:
    ...
    elif k == ord(' '): # shoot with space
      blast_dist = 0.5 # positive value 'sets it off'
      # translation needs 4x4 matrix and x,y,z,w with w set to 1.0
      tip_pt = np.dot(laser_gun.MRaw.T, [0.0, 0.3, 1.0, 1.0])[:3] # gun 'pointing' in +ve z direction
      but_pt = np.dot(laser_gun.MRaw.T, [0.0, 0.3, 0.0, 1.0])[:3] 
      aim_vec = tip_pt - but_pt # because translated need to do this
      # there are some utility functions in the Camera class
      rot_matrix = camera.matrix_from_two_vecors(np.array([0.0, 0.0, 1.0]), aim_vec)
      rot_euler = camera.euler_angles(rot_matrix)
      laser_tip.rotateToX(-rot_euler[0]) # unclear why x and y need to be -ve
      laser_tip.rotateToY(-rot_euler[1])
      laser_tip.rotateToZ(rot_euler[2])
...
```

paddywwoof, 2018-03-20:
P..S. like this (apropos ring 'tweening' and redrawing system) It's very hacky but gives you the kind of idea and would let you test the drawing load.

```
...
# rings
ring = pi3d.Model(file_string=MODEL_PATH.format('straight_ring_2.obj'))
empty = pi3d.Triangle(corners=((0,0),(1,1),(-1,1)))
empty.add_child(ring)
ring.set_shader(shader)
ring.set_fog((0.0, 0.0, 0.0, 0.0), 350.6) # from v2.22 you can change the fog start distance
ring_list = [[0, -2, 50, 10, 5]]
...
while display.loop_running():
  for i in range(6):
    if i >= len(ring_list):
      ring_list.append(ring_list[-1][:])
      ring_list[i][0] -= 50.0 * math.sin(math.radians(ring_list[i][1]))
      ring_list[i][1] -= 50.0 * math.sin(math.radians(ring_list[i][0]))
      ring_list[i][2] += 50.0
      ring_list[i][3] += random.random() * 20.0 - 10.0
      ring_list[i][3] = min(15.0, max(-15.0, ring_list[i][3]))
      ring_list[i][4] += random.random() * 30.0 - 15.0
      ring_list[i][4] = min(20.0, max(-20.0, ring_list[i][4]))
    r = ring_list[i] # short cut for brevity
    r[2] -= 0.1
    r[3] *= r[2] / (r[2] + 0.2)
    r[4] *= r[2] / (r[2] + 0.2)
    empty.positionX(r[0])
    empty.positionY(r[1])
    empty.positionZ(r[2])
    empty.rotateToX(r[3])
    empty.rotateToY(r[4])
    empty.rotateToZ(r[2] * 5.0)
    empty.draw()
    if ring_list[0][2] < 0:
      # check contact
      print(ring_list[0][3:5])
      ring_list = ring_list[1:]
```

scottalmond, 2018-03-21:
Pure awesome.  I'm going to cycle back to this this weekend and digest all this, I deeply appreciate all your help and examples, it's quite a bit to take in.

One idea I was idly thinking about today was using add_child to define objects relative to one another (ex one triangle to another) to make the math easy (this ring center is some translation and rotation away in the local variable space, and the next ring has nearly the same transform, and the next...), then using the back calculation of the object's position/rotation in the global reference frame per your previous post, then breaking that linkage via the remove_child concept, and then re-positioning the triangle back in the same spot, but in the global reference frame rather than linked to another object.  That way I can add rings ahead and remove them behind as needed (no interdependence), I can relate (add_child) the pod to the nearest ring, all objects maintain a predictable global geometry (lasers would fly straight with minimal coding), etc.

> empty.draw()
Is that needed?  If the intent is to not show the empty, why just not render it?

paddywwoof, 2018-03-21:
Yes, interesting idea. I might check out the efficiency of rendering the whole scene as a tree hierarchy. That's the way game engines such as UE, unity, etc do it. They also use quaternions to hold rotations separate from scaling and translation matrices. However this necessitates a fair bit extra processing which I think would hit the python based pi3d quite hard... But there might be a clever way round it, I will have a think.

The way the parent child system works is that when you call parent.draw() it calculates a transformation matrix using T1, Rx, Ry, Rz, S, T2 then passes that to its children etc etc in a recursive way. The resultant transformation matrix (absolute only) is then held by each shape (as MRaw) until the next call to draw(). Maybe there is a case for either marking Shapes as empty to save a little bit of rendering or for splitting out the matrix calculation element of draw().

paddywwoof, 2018-03-21:
Doesn't seem to have a significant impact on processing efficiency... but I'm not convinced it's simpler to code...
I will post the full listings online as it might be getting a bit hard to reconstruct them from these diffs!
```
...
def shallow_clone(shape):
  state = shape.__getstate__()
  clone = pi3d.Triangle() # just saves having to feed in all init arguments, any shape will do
  clone.__setstate__(state) # as everything is overwritten here
  return clone
...
# scene - empty to hold objects
scene = pi3d.Triangle(corners=((0,0),(1,1),(-1,1)))
...
laser_tip.set_alpha(0.0)
...
# rings
ring = pi3d.Model(file_string=MODEL_PATH.format('straight_ring_2.obj'))
empty = pi3d.Triangle(corners=((0,0),(1,1),(-1,1)), z=50.0)
ring.set_shader(shader)
ring.set_fog((0.0, 0.0, 0.0, 0.0), 350.6)
empty.add_child(ring)
ring_list = [empty]
for i in range(5):
  next_empty = shallow_clone(empty)
  next_empty.rotateIncX(random.random() * 40.0 - 20.0)
  next_empty.rotateIncY(random.random() * 40.0 - 20.0)
  next_ring = shallow_clone(ring)
  next_empty.children = [next_ring]
  ring_list[-1].add_child(next_empty)
  ring_list.append(next_empty)

scene.add_child(pod)
scene.add_child(asteroid)
scene.add_child(empty)
scene.add_child(laser_tip)
...
while display.loop_running():
  scene.draw()

  ring_list[0].translateZ(-0.1)
  rx, ry = ring_list[0].unif[3:5]
  ring_list[0].rotateToX(rx * 0.995)
  ring_list[0].rotateToY(ry * 0.995)

  if ring_list[0].z() < 0:
    # check clash
    r_swap = ring_list[0]
    scene.children.remove(r_swap)
    ring_list[-1].add_child(r_swap)
    ring_list = ring_list[1:] + [r_swap]
    r_swap.children.remove(ring_list[0])
    r_swap.positionZ(50.0)
    r_swap.rotateIncX(random.random() * 40.0 - 20.0)
    r_swap.rotateIncY(random.random() * 40.0 - 20.0)
    
    scene.add_child(ring_list[0])

  for r in ring_list:
    r.children[0].rotateIncZ(1.0)

  asteroid.rotateIncX(1.5)
...
  if blast_dist > 0.0:
    laser_tip.set_alpha(0.8) # use alpha to switch render on and off
    laser_tip.position(*(tip_pt + aim_vec * blast_dist))
    blast_dist += 5.0
    if blast_dist > RANGE:
      blast_dist = 0.0
      laser_tip.set_alpha(0.0)

```

paddywwoof, 2018-03-23:
Added methods to Shape in develop branch. I will issue v2.23 quite soon.
rotate_to_direction()
transform_direction()
shallow_clone()
https://github.com/tipam/pi3d/commit/74fcc0e0cae369059a2ad5e3830e0f5ced90b9fb
and the examples I put on github.com/paddywwoof/ above are altered to used these new methods.

closed 2018-03-23 by paddywwoof

scottalmond, 2018-03-28:
@paddywwoof the changes look awesome and I'm eager to get started.  Do you have an idea for how soon v2.23 will be released?  I'd favor doing a pip3 upgrade rather than puzzling out how to create and use a parallel install of the pi3d development branch

paddywwoof, 2018-03-28:
trying to get this out today. Just need to run tests on windows - but there should be anything broken from these changes.

paddywwoof, 2018-03-29:
v2.23 should be picked up from pypi (i.e. using pip) from 29th Mar

paddywwoof, 2018-04-09:
@scottalmond noticed that sometimes one of the methods gives an error with python2's class definitions (doesn't like None as an instance of Camera) so I've fixed this and issued a bug-fix release v2.24. You should be able to ``sudo pip install pi3d --upgrade``