tipam/pi3d #218: exit

tekcomm, 2018-03-28:


rec, 2018-03-28:
We'll need more information than this to continue!  :-)

tekcomm, 2018-03-30:
root@AMD-rocm-rippa:/home/guru# dmesg | grep amd
[    1.105708] [drm] amdgpu kernel modesetting enabled.
[    1.125128] fb: switching to amdgpudrmfb from EFI VGA
[    1.127266] amdgpu 0000:01:00.0: Invalid PCI ROM header signature:
expecting 0xaa55, got 0xffff
[    1.127338] amdgpu 0000:01:00.0: BAR 2: releasing [mem
0xe0000000-0xe01fffff 64bit pref]
[    1.127341] amdgpu 0000:01:00.0: BAR 0: releasing [mem
0xd0000000-0xdfffffff 64bit pref]
[    1.127363] amdgpu 0000:01:00.0: BAR 0: assigned [mem
0x2000000000-0x21ffffffff 64bit pref]
[    1.127370] amdgpu 0000:01:00.0: BAR 2: assigned [mem
0x2200000000-0x22001fffff 64bit pref]
[    1.127394] amdgpu 0000:01:00.0: VRAM: 8192M 0x000000F400000000 -
0x000000F5FFFFFFFF (8192M used)
[    1.127396] amdgpu 0000:01:00.0: GTT: 256M 0x0000000000000000 -
0x000000000FFFFFFF
[    1.127445] [drm] amdgpu: 8192M of VRAM memory ready
[    1.127447] [drm] amdgpu: 8192M of GTT memory ready.
[    1.865562] [drm] Initialized amdgpu 3.23.0 20150101 for 0000:01:00.0 on
minor 0
[    1.865596] amdgpu 0000:05:00.0: enabling device (0000 -> 0003)
[    2.940308] amdgpu 0000:05:00.0: BAR 2: releasing [mem
0x2ff0000000-0x2ff01fffff 64bit pref]
[    2.940311] amdgpu 0000:05:00.0: BAR 0: releasing [mem
0x2fe0000000-0x2fefffffff 64bit pref]
[    2.940347] amdgpu 0000:05:00.0: BAR 0: assigned [mem
0x2400000000-0x25ffffffff 64bit pref]
[    2.940358] amdgpu 0000:05:00.0: BAR 2: assigned [mem
0x2300000000-0x23001fffff 64bit pref]
[    2.940392] amdgpu 0000:05:00.0: VRAM: 8192M 0x000000F400000000 -
0x000000F5FFFFFFFF (8192M used)
[    2.940394] amdgpu 0000:05:00.0: GTT: 256M 0x0000000000000000 -
0x000000000FFFFFFF
[    2.940406] [drm] amdgpu: 8192M of VRAM memory ready
[    2.940408] [drm] amdgpu: 8192M of GTT memory ready.
[    3.658624] [drm] Initialized amdgpu 3.23.0 20150101 for 0000:05:00.0 on
minor 1
root@AMD-rocm-rippa:/home/guru# cl
clear          clear_console  clinfo
root@AMD-rocm-rippa:/home/guru# cl
clear          clear_console  clinfo
root@AMD-rocm-rippa:/home/guru# clinfo
Number of platforms:  1
  Platform Profile:  FULL_PROFILE
  Platform Version:  OpenCL 2.1 AMD-APP (2527.3)
  Platform Name:  AMD Accelerated Parallel Processing
  Platform Vendor:  Advanced Micro Devices, Inc.
  Platform Extensions:  cl_khr_icd cl_amd_event_callback
cl_amd_offline_devices

  Platform Name:  AMD Accelerated Parallel Processing
Number of devices:  2
  Device Type:  CL_DEVICE_TYPE_GPU
  Vendor ID:  1002h
  Board name:  Radeon RX 570 Series
  Device Topology:  PCI[ B#1, D#0, F#0 ]
  Max compute units:  32
  Max work items dimensions:  3
    Max work items[0]:  1024
    Max work items[1]:  1024
    Max work items[2]:  1024
  Max work group size:  256
  Preferred vector width char:  4
  Preferred vector width short:  2
  Preferred vector width int:  1
  Preferred vector width long:  1
  Preferred vector width float:  1
  Preferred vector width double:  1
  Native vector width char:  4
[... 199 lines cut ...]
root@AMD-rocm-rippa:/home# cd guru/
root@AMD-rocm-rippa:/home/guru# ls
add2bashrc         linux-headers-4.16.0-041600rc
7_4.16.0-041600rc7.201803252030_all.deb            text-mode-fonts
clean-myself.bash  linux-headers-4.16.0-041600rc7
-generic_4.16.0-041600rc7.201803252030_amd64.deb  text-mode-installer
fontlist           linux-image-4.16.0-041600rc7-
generic_4.16.0-041600rc7.201803252030_amd64.deb    text-mode-menu
l                  r8168-dkms_8.045.08-2_all.deb
                       truncate-log

Attached is the kernel version to try
​
 linux-headers-4.16.0-041600rc7-generic_4.16.0-0...
<https://drive.google.com/file/d/1hdaPRU02W8GLP5e1ib35Fzb-FXtX5w9t/view?usp=drive_web>
​​
 linux-headers-4.16.0-041600rc7_4.16.0-041600rc7...
<https://drive.google.com/file/d/1qKAcyZx52z8c1MfeZz-gIX7JxSacFemv/view?usp=drive_web>
​​
 linux-image-4.16.0-041600rc7-generic_4.16.0-041...
<https://drive.google.com/file/d/1bstQ1wHZYDRh3Xszt3P7KuwmJ22jsBbb/view?usp=drive_web>
​​
 r8168-dkms_8.045.08-2_all.deb
<https://drive.google.com/file/d/17hQJu6EpTaJpoqGAK2artxx3Ie7NA7O-/view?usp=drive_web>
​

rec, 2018-03-30:
By "more information" I mean that you haven't actually told us what your issue _is._

tekcomm, 2018-03-30:
Hahhaha, This is called sleep derpervation, I had a issue with the I7 Gen3 support it was a issues related to grub.

tekcomm, 2018-03-30:
I posted the fix for other intel board users I tested it up to the 4.17 prerelease

tekcomm, 2018-03-30:
I am so used to being in terminal mode after 40 years I was 1/2 sleep and typed, You guested it exit LFAO

tekcomm, 2018-03-30:
Sorry bout that, I was up for like 30+ hours coding and face-palmed
https://github.com/RadeonOpenCompute/ROCm/issues/361 building a live
userland in 2 x 30+ hour coding benders lol

closed 2018-03-30 by tekcomm

rec, 2018-03-30:
You will rot your brain that way!!  :-D

tekcomm, 2018-03-30:
I looked at your code, I can tell you don't practice what ya preach hahahha

tekcomm, 2018-03-30:
Im like oh look a zombie Apocalypse, click click click,  lol