tipam/pi3d #219: Nearfield Fog

scottalmond, 2018-04-10:
I note there is a fog effect for items far away, but is there a near-field equivalent that operates in the opposite way (as items get closer, they vanish)?  I'm trying to replicate an effect as seen with the asteroids 15:12 to 15:14 here (they vanish as they get very close to the camera):
https://www.youtube.com/watch?v=Er0AzrrjrJI&feature=youtu.be&t=15m12s
My current notion is to twiddle the set_alpha value for objects close to the camera as a function of distance, but I'm wondering if there may already be something in the library that does this

paddywwoof, 2018-04-10:
Hi, there are two ways of doing this (apart from twiddling Shape.alpha which might not not be too inefficient really, but requires all the camera-to-Shape distances to be calculated each frame, unless the  camera always points +ve z).
1. Use the near plane i.e.
`camera = pi3d.Camera(lens=(15.0, display.INSTANCE.far, display.INSTANCE.fov, display.INSTANCE.width / float(display.INSTANCE.height)))`
which would be quite efficient but gives a sharp cutoff so might look weird.
2. Tweak the shader. You can just copy uv_light.fs and uv_light.vs (or whichever you're using) from pi3d/pi3d/shaders to the same folder or sub folder of your python file then
`shader = pi3d.Shader('myshaderdir/uv_light_nearfade')`
then alter the alpha like
`  gl_FragColor.a *= unif[5][2] * smoothstep(12.0, 18.0, dist);`
(dist is a varying value defined in `#include` at the top, you could pass uniform values for 12.0 and 18.0 using the Shape.set_custom_data()) this gives a smooth fade. NB you would need to draw any fading Shape after anything you would be able to see through it.

closed 2018-04-10 by paddywwoof

paddywwoof, 2018-04-10:
TLDR; Looking at this a bit more I've found a couple of issues that need thinking about and fixing. The last four lines that move the camera need to be done before any of the Shape.draw() calls.

In order to work out what what part of the reflection image should be visible from the camera, given the position and normal vector of a specific fragment, there is a uniform variable set on each Shape whenever the camera is moved. This is done in the Shape.draw() function so the camera needs to be moved before the draw() calls otherwise it won't be picked up. The pi3d shaders all use the length of the vector from the camera to the vertex as the depth. This means that things nearer the middle of the screen are closer to the camera than things at the edges, which might be good, but might not really make much difference as it's only really relevant for very close things. Alternatively:
1. vertex shaders could use gl_Position[2] which would work even if the camera matrix was updated after the Shape.draw()
2. fragment shaders could use gl_FragCoord[2] but this is very non-linear and compressed into the range 0.0 at the near plane to 1.0 at the far plane

PS it's also rather messy to change the code so that the objects always get drawn from furthest to nearest.