tipam/pi3d #223: TextBlock colouring.set_colour incongruent results

frankgould, 2018-04-26:
I am trying to set the colors for text in a TextBlock but I get unexpected results that I cannot figure out why it is happening.  I use a color lookup table to get the RGB values to use in the colouring.set_colour call but what shows on the screen does not always match.  In the log, I save what the values are sent to the call and some examples appear below.

Set day/date/time RGB to yellow, white, magenta: 
day_text colour(yellow): [1.0, 1.0, 0.0, 0.75] Displayed white (should be yellow)
date_text colour(white): [1.0, 1.0, 1.0, 0.75] Displayed yellow (should be white)
time_text colour(magenta): [1.0, 0.0, 1.0, 0.75] Displayed magenta (correctly)

Set RGB to orange, pink, yellow:
day_text colour(orange): [1.0, 0.64, 0.0, 0.75] Displayed pink
date_text colour(pink): [1.0, 0.71, 0.75, 0.75] Displayed yellow
time_text colour(yellow): [1.0, 1.0, 0.0, 0.75] Displayed white

Set RGB to blue, red, green:
day_text colour(blue): [0.0, 0.0, 1.0, 0.75] Displayed blue (correctly)
date_text colour(red): [1.0, 0.0, 0.0, 0.75] Displayed red (correctly)
time_text colour(green): [0.0, 1.0, 0.0, 0.75] Displayed cyan

The code I use to setup the text management is as follows:

    caption0_color_R = 1.0
    caption0_color_G = 1.0
    caption0_color_B = 1.0
    font_path = os.path.abspath(os.path.join('/usr/share/fonts/truetype/freefont/', 'fonts', 'FreeSerifBold.ttf'))
    font_colour = (255, 255, 255, 255)
    pointFont = pi3d.Font(font_path, font_colour, codepoints=list(range(32,128)))
    caption_text = pi3d.PointText(pointFont, CAMERA, max_chars=250, point_size=84)
    caption_0_txt = pi3d.TextBlock(-25, -510, 0.2, 0.0, 100, data_obj=clock_caption_object, attr="strCaption0",
            text_format="{:s}", size=font_size_list[captions_fontsize], spacing="F", space=-0.05, 
            colour=(caption0_color_R, caption0_color_G, caption0_color_B, caption0_set_alpha), justify=0.5)
    caption_text.add_text_block(caption_0_txt)
    caption_0_txt.colouring.set_colour([caption0_color_R, caption0_color_G, caption0_color_B, caption0_set_alpha])
    color=[caption0_color_R, caption0_color_G, caption0_color_B, caption0_set_alpha]
    LOGGER.info('caption_0_txt.colouring.set_colour(' + str(captions_color_fg) + '): ' + str(color) + '\r')
    caption_1_txt.colouring.set_colour([caption1_color_R, caption1_color_G, caption1_color_B, caption1_set_alpha])
    color=[caption1_color_R, caption1_color_G, caption1_color_B, caption1_set_alpha]
    LOGGER.info('caption_1_txt.colouring.set_colour(' + str(captions_color_fg) + '): ' + str(color) + '\r')

Please let me know what I can do to isolate this problem to be corrected in whatever way I can.  I've tried String but it failed to work on my DISPLAY with other planes and text.

paddywwoof, 2018-04-26:
Hi, that's a bug - well spotted. It was only copying over the red and green values. I've pushed up a fix to the develop branch. https://github.com/tipam/pi3d/commit/431ebb68a9e9dd8642ed60372ebe7f729340e75f#diff-984d894877fe1219424c0563d7ec47b4 and will put a new version up on pypi.python.org so you can pip upgrade tomorrow.

paddywwoof, 2018-04-27:
v2.25 has the, hopefully, correct behavior. It will update RGB if a three tuple or array is passed to set_colour() or RGBA if a four tuple. Alternatively, or at the same time alpha can be updated with an additional argument.

closed 2018-04-27 by paddywwoof

frankgould, 2018-04-27:
Yes paddywwoof it behaves correctly. I’ve been working on it all day to confirm but I found other anomalies in my code that slowed me down. Thank you.