tipam/pi3d #224: isotropic Display for 1920x1200 monitors

mrmsbarnes, 2018-05-15:
Something believes my 1920 x 1200 monitor has only 1080 units of height.  Spheres are stretched to eggs. I'd like to see correct display of 3D objects.

mrmsbarnes, 2018-05-16:
Yes, we are only talking RPi (thank you, though, for investing so much work for multi-platform).  When I use a 1920x1080 monitor there is no problem (ViewSonic TD2220).  I see anisotropic stretching on my 1920x1200 monitor (Asus VS24A).  I think you are suggesting there is a disagreement between bcm and x11 for this case.  Do you have means to test on a 1920x1200 monitor?

mrmsbarnes, 2018-05-16:
Sorry, maybe there is a disagreement between bcm and pygame.  Is x11 even used on RPi?

paddywwoof, 2018-05-16:
I've not managed to get x11 or pygame to work on the RPi (but haven't tried too hard either). I've seen a couple of comments on forums about getting the pi or monitor to set themselves up correctly, various suggestions for using different HDMI numbers.

To check what the broadcom driver thinks the monitor is you can type this in a python terminal
```
$ python3
>>> import pi3d
>>> from pi3d.constants import bcm
>>> import ctypes
>>> bcm.bcm_host_init()
>>> w = ctypes.c_int()
>>> h = ctypes.c_int()
>>> bcm.graphics_get_display_size(0, ctypes.byref(w), ctypes.byref(h))
>>> print(w, h)
```
If you can't get the monitor detection to work properly you can quite easily modify the value used in pi3d by changing it just after the Display object has been created:
```
display = pi3d.Display.create()
display.width = 1920
display.height = 1200
```
Ideally you would change the fourth value in the [lens array that you can optionally pass to the Camera](http://pi3d.github.io/html/pi3d.html#module-pi3d.Camera) which defines the aspect ratio, however the orthographic camera doesn't have a lens and uses display.width and display.height directly.

Hope this solves your problem

closed 2018-05-16 by paddywwoof