tipam/pi3d #225: pi3d and streaming video crash

frankgould, 2018-05-25:
I have installed and successfully run picam video streaming across node-rtmp-rtsp-server using coffee. I tested this for months while I migrated my simple slideshow app to use pi3d instead of kivy. About a month ago, my app started crashing the video streaming and generated the following error message:

`warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.`

I experienced the same result using only the pi3d Shapes.py demo app while the video streaming was working and it too crashed.  Below are the log entries from my app that displays GPU free memory every 15 seconds.

```
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:01:31}
Warning NEXT: LOW GPU memory=8.6K Date/Time= {2018-05-20 09:01:46}
NEXT: GPU memory=354M free memory in 70 free block(s) Date/Time= {2018-05-20 09:02:01}
NEXT: GPU memory=354M free memory in 68 free block(s) Date/Time= {2018-05-20 09:02:16}
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:02:31}
NEXT: GPU memory=354M free memory in 70 free block(s) Date/Time= {2018-05-20 09:02:46}
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:03:01}
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:03:16}
Warning NEXT: LOW GPU memory=8.5K Date/Time= {2018-05-20 09:03:31}
NEXT: GPU memory=354M free memory in 73 free block(s) Date/Time= {2018-05-20 09:03:46}
NEXT: GPU memory=354M free memory in 68 free block(s) Date/Time= {2018-05-20 09:04:01}
NEXT: GPU memory=354M free memory in 68 free block(s) Date/Time= {2018-05-20 09:04:16}
NEXT: GPU memory=354M free memory in 70 free block(s) Date/Time= {2018-05-20 09:04:31}
NEXT: GPU memory=354M free memory in 72 free block(s) Date/Time= {2018-05-20 09:04:47}
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:05:02}
NEXT: GPU memory=354M free memory in 69 free block(s) Date/Time= {2018-05-20 09:05:17}
NEXT: GPU memory=354M free memory in 72 free block(s) Date/Time= {2018-05-20 09:05:32}
NEXT: GPU memory=354M free memory in 71 free block(s) Date/Time= {2018-05-20 09:05:47}
NEXT: GPU memory=354M free memory in 72 free block(s) Date/Time= {2018-05-20 09:06:02}
NEXT: GPU memory=354M free memory in 73 free block(s) Date/Time= {2018-05-20 09:06:17}
NEXT: GPU memory=354M free memory in 70 free block(s) Date/Time= {2018-05-20 09:06:32}
NEXT: GPU memory=354M free memory in 68 free block(s) Date/Time= {2018-05-20 09:06:47}
Warning NEXT: LOW GPU memory=5.1K Date/Time= {2018-05-20 09:07:02}
```

Here is the code that generates the above log messages:
```
      p1 = Popen(["sudo vcdbg reloc |grep 'free memory'"], stdout=PIPE, shell=True)
      gpu_temp = p1.communicate()[0]
      Popen.kill
      gpu_temp = gpu_temp.rstrip()
      split_gpu_temp=gpu_temp.split()
      if 'M' not in split_gpu_temp[0] and 'G' not in split_gpu_temp[0]:
        if mp_frame.debug_enabled: LOGGER.warning('Warning update: LOW GPU memory=' + str(split_gpu_temp[0]) + ' Date/Time= {' + str(datetime.now())[0:19] + '}\r')
      else:
        if mp_frame.debug_enabled: LOGGER.info('update: GPU memory=' + str(gpu_temp) + ' Date/Time= {' + str(datetime.now())[0:19] + '}\r')
```
 I searched for a free cache method thinking that might help but I could not find a solution. What is the best way to determine what is causing the GPU memory to be consumed so much and how to balance the load?

paddywwoof, 2018-05-28:
Hi, it's hard to say (without knowing anything about node-rtmp-rtsp-server using coffee or seeing your code) but it is odd that it worked ok previously then stopped. Did anything change at the time it started crashing - have you tried increasing or decreasing the GPU memory allocaiton in raspi-config to see if that has any impact (Though it looks like there's already a lot alocated, is that the max value)?

Are you creating new Textures or are you using Texture.update_ndarray() as in VideoWalk or are you using a system like @swehner's foos https://github.com/swehner/foos ?

If you create lots of Textures then the memory ought to be released by the python GC as there is a Texture.__del__() method that adds a reference to the OpenGL texture and the Display._tidy() method that runs each frame (when there is a tidy needed) should release the memory. But the the GC isn't guaranteed to run any specific time so there might be delays. You could probably get the gl ref to the Texture you want to release and explicity call glDeleteTextures, something along the lines of
        t = my_texture_to_delete._tex
        pi3d.opengles.glDeleteTextures(1, ctypes.byref(t))
which might cause other errors as the automatic tidying tries to delete the texture as well (but seems to work on a very simple test)

frankgould, 2018-05-30:
Thanks paddywwoof for the suggestions and I wish I knew what changed in my code to cause this problem. It could be when I migrated to pi3d because that's what appears to be causing the Shapes.py code from causing the same problem as my app (see my original post).

I have tried to increase memory with mixed results. Per [RPi online doc](https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md), I tried 1024 and 944 which both locked up my [RPi 3B that supposedly has 1GB RAM](https://www.amazon.com/gp/product/B01N9YSKEU/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1). I'm testing 656MB, which I found someone suggested that in another post.

As I was searching for your suggestion, I found that the python garbage collection (gc.collect()) could make a difference which I have added back to my code to see if there is any improvement with that solution. I wanted to reply and thank you for your suggestion after I had some time to research this and respond.

I'll report back in a few days after testing these changes to see what results I get.

frankgould, 2018-05-31:
Turns out it doesn't take but a day to find it failed at 656MB that appears to be more than RPi can handle because it crashed due to some network call with 'out of resources' error. It also appears gc.collect() does not have any improvement but I will continue to test my app with 512MB which has not crashed in the past like 656MB did.

paddywwoof, 2018-05-31:
Might be a stupid question but have you tried reducing --gopsize or enabling variable frame rate (if that's in any way relevant or possible)