tipam/pi3d #227: Display.resize() corrupts layer

KenT2, 2018-10-25:
The problem is that Display.resize() does not have a layer parameter or use the layer fromm create(), therefore DisplayOpenGL.resize() uses its default layer=0 parameter.

If it works it would be good to add layer to resize so layers can be changed dynamically, otherwise use the value from Display.create()

paddywwoof, 2018-10-26:
@KenT2 I've added layer argument to DisplayOpenGL.resize and made Display.resize pass the self.layer value which is now saved in create() and in a new function change_layer()
This is now in the develop branch but if you need it for something you're working on I can put it in master/pypi and bump the version number.
https://github.com/tipam/pi3d/commit/5058b63364a8e8ca71f6e83d8aa7594ea828b007

paddywwoof, 2018-10-28:
Ken, I remember you taking about that before (best not to think how many  years!) If you've got any questions, or suggestions for improving compatibility etc please get in touch, I'd be glad to help. patrick at eldwick co uk