tipam/pi3d #228: Issues loading texture while drawing

mikeyy, 2018-11-07:
After around 20 times loading a new texture over an existing being drawn assigned variable/texture, either the texture fails to load completely or one of the models loses orientation and doesn't register set positions. Am I loading textures wrong? Am I encountering a bug? How can I avoid this issue?

paddywwoof, 2018-11-08:
It sounds like a memory leak type of bug. At one point there was a GPU side memory leak in pi3d (not the whole pixel memory, just the OpenGL identifiers) but this should have been fixed - but it wouldn't show up after only 20 images. The pi3d_demo slideshows and pictureframe reload textures so they might be worth looking at.

Do you have code available in a public repo for me to peer at?

paddywwoof, 2018-11-09:
Hi, just remembered that I did concoct https://github.com/tipam/pi3d/blob/master/experiments/LeakTest.py (so already in your pi3d/experiments directory) to test gpu memory leakage. It might help you track down whether that is the root of your problem.

Let me know what you find as, obviously, this needs to be sorted or clarified for other users of pi3d.

paddywwoof, 2018-12-13:
hi @mkeyy, glad to hear that you sorted it out and it's not something I need to fix in pi3d. There are a couple of things worth pointing out in the code you linked to that might be relevant, especially if you find you have to wait for something to complete before moving on.

One thing people mention coming to python from other languages is that it's often quite slow. But the fact is that it's capable of being much faster - even comparable with C or Rust - because the processing power of numpy is **_very_** sophisticated! That's why I ended up using it for the array buffer data in pi3d. The extra irony is that using normal python on numpy arrays is slower than on standard python lists because of the conversion overhead. And a last point: because python can return tuples you don't need to set a tmp variable - you can juggle the swapping in one line!

So using the timer sample below I get:
```
 100    10000 number of vertices
0.23     22.2 ms - your code
0.11     10.4 ms - your code but using numpy indexing i.e. [a][b] -> [a,b]
0.10     10.5 ms - swapping a tuple with a tuple, numpy indexing
0.005     0.15 ms - numpy one liner!
```

```
import timeit

setup = '''
import numpy as np
a = np.random.random((10000, 7))
'''

funcs = ['''
for i in range(a.shape[0]):
    tmp = a[i][1]
    a[i][1] = a[i][2]
    a[i][2] = -tmp
    tmp = a[i][4]
    a[i][4] = a[i][5]
    a[i][5] = -tmp
''', '''
for i in range(a.shape[0]):
    tmp = a[i,1]
    a[i,1] = a[i,2]
    a[i,2] = -tmp
    tmp = a[i,4]
    a[i,4] = a[i,5]
    a[i,5] = -tmp
''', '''
for i in range(a.shape[0]):
    a[i,1], a[i,2] = a[i,2], -a[i,1]
    a[i,4], a[i,5] = a[i,5], -a[i,4]
''', '''
a[:,1], a[:,2] = a[:,2], -a[:,1]
a[:,4], a[:,5] = a[:,5], -a[:,4]
''']
for f in funcs:
    print(timeit.timeit(f, setup, number=100))
```

paddywwoof, 2018-12-13:
Hi, @mikeyy - as a follow on but different aspect (so I will make it a different post) If you have issues with rotated frame of reference making things misbehave then a good solution is to make the shape a child of another shape. You can then rotate the parent shape to point the child axes in the right direction. This is how, for instance, the ball in the pong demo https://github.com/pi3d/pi3d_demos/blob/master/Pong.py#L53 rolls about a horizontal axis at right angles to its direction of travel.

mikeyy, 2018-12-19:
Hey @paddywwoof,

I was able to successfully replace the for loop with the numpy one-liner you provided, and it's working wonders. I'm going to experiment with numpy for the next few days and see if I can apply these optimizations elsewhere in the code before production. Thankfully, the rotation issue hasn't surfaced again.

Thanks a bunch, you've helped expand my knowledge even more.