tipam/pi3d #229: Using filters with Stereocam

grindash, 2019-02-12:
Hi,
There seems to be a conflict between the Stereocam command and the PostProcess command because they use the same type of fonctionnality.
I can switch to a regular camera with Camera.instance for the filter but I lose the double screen if I do that.
I also tried to use 2 different cameras that I switch by an input but it's not working : the first StereoCam will work perfectly but the second one will be a freezed image (and it's not because of my filter: if I change the order of the lines the one working fine will always be the first line).
```
self.CameraFiltre=pi3d.StereoCam(separation=self.hmd_eye_seperation, interlace=0, shader="/home/pi/pi3d_demos/shaders/filter_plant")
self.CAMERA = pi3d.StereoCam(separation=self.hmd_eye_seperation, interlace=0, shader=shader_name)
```
Is there a way to update the shader of a single StereoCam ?

paddywwoof, 2019-02-12:
hi, I'm sure it's possible to combine the functionality but almost certainly need revised shader at least. If the filter can be run on the combined left+right image the I suspect it will be easier than running filter separately on each half. I will look at this later

paddywwoof, 2019-02-12:
Hi, I've just had a quick go and it seems to work mainly ok. I used the existing ForestStereo.py demo and just changed the lines
```
CAMERA = pi3d.StereoCam(separation=-0.5, interlace=0, shader="/home/pi/pi3d_demos/shaders/filter_outline")
for s in CAMERA.sprites:
  s.unif[45:47] = [DISPLAY.width, DISPLAY.height]
  s.unif[48:51] = [1.0, 0.5, 0.2]
```
but I'm away from RPi at the moment so running on laptop. Can you post a copy of your filter shader so I can try it here? Does it depend on uniform variables passed in Shape.unif of Buffer.unib arrays?

Paddy

grindash, 2019-02-13:
Firstly, thanks for working so quickly.
Secondly here's my filter, it's mostly a modification of the "color_dist" filter:
filter_plant.fs
```
precision mediump float;
varying vec2 uv;
uniform sampler2D tex0;
void main(void){
  vec4 color = texture2D(tex0, uv);
  gl_FragColor.r = color.r;
  gl_FragColor.g = 0.;
  gl_FragColor.b = color.b;
  gl_FragColor.a = 1.;
;
}
```

filter_plant.vs
```
precision mediump float;
attribute vec3 vertex;
attribute vec2 texcoord;
uniform mat4 modelviewmatrix[2]; // 0 model movement in real coords, 1 in camera coords
uniform vec3 unib[4];
//uniform vec2 umult, vmult => unib[2]
//uniform vec2 u_off, v_off => unib[3]
varying vec2 uv;
void main(void) {
  uv = texcoord * unib[2].xy + unib[3].xy;
  uv.y = 1.0 - uv.y;
  gl_Position = modelviewmatrix[1] * vec4(vertex,1.0);
}
```
Yes I need to add something to flip this filter upside-down but I have'nt worked on that for now.

paddywwoof, 2019-02-13:
@grindash had a quick look at this and it seemed to work ok on this laptop using your filter but I commented out the line in your vertext shader flipping the texture (`//uv.y = 1.0 - uv.y;`) Also in ForestStereo.py I only set the width and height in the two sprites as you don't pass any other uniform values to the shader. [see](https://drive.google.com/open?id=11UQTbXgQd02zAl-W3cMSxWQjT86Bm4MN) Try this on your RPi and let me know what happens.

PS if you want a different effect for the left and right image then you would need to set the shader on each sprite rather than using the default mechanism done for you by StereoCam

grindash, 2019-02-14:
Well I completly forgot the shaders wasn't in python and tried to comment with "#", that's why it didn't work...
I tried adding the "sprites" line but it didn't resolve my problem. I think I wasn't clear on what I wanted to do : 
When I press a key, I want to switch between the "uv_flat" filter and my filter
```
def render_stereo_scene(self, render_callback):
	if self.PlantFilter == True:
		self.CameraFiltre.move_camera(position=self.camera_position, rot=self.camera_rotation[1], tilt=self.camera_rotation[0], roll=-self.camera_rotation[2])
	else:
		self.CAMERA.move_camera(position=self.camera_position, rot=self.camera_rotation[1], tilt=self.camera_rotation[0], roll=-self.camera_rotation[2])
	
	if self.PlantFilter == True:
		self.CameraFiltre.start_capture(0)
		render_callback() #list of elements to draw
		self.CameraFiltre.end_capture(0)
		self.CameraFiltre.draw()
	else:
		self.CAMERA.start_capture(0)
		render_callback()
		self.CAMERA.end_capture(0)
		self.CAMERA.draw()
```
With this, when I switch between the camera, the shader change but one will be stuck like if the "move_camera" command wasn't working.
I've also tried this:
```
if self.PlantFilter == True:
	self.CameraFiltre.move_camera(position=self.camera_position, rot=self.camera_rotation[1], tilt=self.camera_rotation[0], roll=-self.camera_rotation[2])
self.CAMERA.move_camera(position=self.camera_position, rot=self.camera_rotation[1], tilt=self.camera_rotation[0], roll=-self.camera_rotation[2])
```
But the camera start spinning eldlessly when the 2 camera are up at the same time.

PS: Sorry for being confusing, english is not my native language and I'm not a developper.

paddywwoof, 2019-02-14:
Yes, I'm constantly swapping # and // when I switch between python and GLSL, rust, C etc!

It's probably worth gradually changing a app and making sure it still works rather than jumping in with something too complicated that might have different problems happening at the same time.

To switch a shader for a Shape.draw() is actually quite easy as it has shader as an argument. However the Shape objects are a bit hidden in the StereoCam object so not quite so difficult but still pretty easy. The first example shows mods to ForestStero to do that. However it's possible to pass arguments to the shader and not swap it. i.e. you pass a Vec3 or RGB multiples so you could easily decrease (or increase!) the shades. I like the latter method better.

using your shader unaltered (apart from commenting out y flip as mentioned above)

```
#!/usr/bin/python
from __future__ import absolute_import, division, print_function, unicode_literals

import math,random

import demo
import pi3d

# Setup display and initialise pi3d
DISPLAY = pi3d.Display.create(w=1200, h=600, frames_per_second=30)
DISPLAY.set_background(0.4,0.8,0.8,1)      # r,g,b,alpha
# yellowish directional light blueish ambient light
pi3d.Light(lightpos=(1, -1, -3), lightcol=(1.0, 1.0, 0.8), lightamb=(0.25, 0.2, 0.3))
CAMERA = pi3d.StereoCam(separation=-0.5, interlace=0, shader="/home/patrick/python/pi3d_demos/shaders/filter_plant")
for s in CAMERA.sprites:
  s.unif[45:47] = [DISPLAY.width, DISPLAY.height]
flatsh = pi3d.Shader("uv_flat") # for shader switching
filtsh = CAMERA.shader # keep a reference to the filter shader to switch back and forth
plant_filter = True

# load shader
shader = pi3d.Shader("uv_bump")
shinesh = pi3d.Shader("uv_reflect")
flatsh = pi3d.Shader("uv_flat")

tree2img = pi3d.Texture("textures/tree2.png")
tree1img = pi3d.Texture("textures/tree1.png")
hb2img = pi3d.Texture("textures/hornbeam2.png")
bumpimg = pi3d.Texture("textures/grasstile_n.jpg")
reflimg = pi3d.Texture("textures/stars.jpg")
rockimg = pi3d.Texture("textures/rock1.jpg")

FOG = ((0.3, 0.3, 0.4, 0.8), 650.0)
TFOG = ((0.2, 0.24, 0.22, 1.0), 150.0)

#myecube = pi3d.EnvironmentCube(900.0,"HALFCROSS")
ectex=pi3d.loadECfiles("textures/ecubes","sbox")
myecube = pi3d.EnvironmentCube(size=900.0, maptype="FACES", name="cube")
myecube.set_draw_details(flatsh, ectex)

# Create elevation map
mapsize = 1000.0
mapheight = 60.0
mountimg1 = pi3d.Texture("textures/mountains3_512.jpg")
mymap = pi3d.ElevationMap("textures/mountainsHgt.png", name="map",
                     width=mapsize, depth=mapsize, height=mapheight,
                     divx=32, divy=32) 
mymap.set_draw_details(shader, [mountimg1, bumpimg, reflimg], 128.0, 0.0)
mymap.set_fog(*FOG)

#Create tree models
treeplane = pi3d.Plane(w=4.0, h=5.0)

treemodel1 = pi3d.MergeShape(name="baretree")
treemodel1.add(treeplane.buf[0], 0,0,0)
treemodel1.add(treeplane.buf[0], 0,0,0, 0,90,0)

treemodel2 = pi3d.MergeShape(name="bushytree")
treemodel2.add(treeplane.buf[0], 0,0,0)
treemodel2.add(treeplane.buf[0], 0,0,0, 0,60,0)
treemodel2.add(treeplane.buf[0], 0,0,0, 0,120,0)
[... 300 lines cut ...]
  if not l_or_k_pressed: #to stop overwriting move_camera() after pressing l
    CAMERA.move_camera((xm, ym, zm), rot, tilt, roll, absolute=False)
  rot, tilt, roll = 0.0, 0.0, 0.0
  myecube.position(xm, ym, zm)
  for i in range(2):
    CAMERA.start_capture(i)
    monument.draw()
    mymap.draw()
    if abs(xm) > 300:
      mymap.position(math.copysign(1000,xm), 0.0, 0.0)
      mymap.draw()
    if abs(zm) > 300:
      mymap.position(0.0, 0.0, math.copysign(1000,zm))
      mymap.draw()
      if abs(xm) > 300:
        mymap.position(math.copysign(1000,xm), 0.0, math.copysign(1000,zm))
        mymap.draw()
    mymap.position(0.0, 0.0, 0.0)
    myecube.draw()
    mytrees3.draw()
    mytrees2.draw()
    mytrees1.draw()
    CAMERA.end_capture(i)
  CAMERA.draw()
```

paddywwoof, 2019-02-14:
PS, I've just looked a bit more carefully at your shader and you don't actually use the Display.width and height so you can take that out of your code. i.e. in the first example above:
```

CAMERA = pi3d.StereoCam(separation=-0.5, interlace=0, shader="/home/pi/pi3d_demos/shaders/filter_plant")
filtsh = CAMERA.shader # keep a reference to the filter shader to switch back and forth
plant_filter = True
```

grindash, 2019-02-15:
Thank you so much, the CAMERA.shader did the trick !

closed 2019-02-15 by grindash