tipam/pi3d #230: Pi3d on ubuntu core

matteoraggi, 2019-05-23:
Hi, is pi3d running on ubuntu core too for raspberry ? and outside raspberry?
Are there any plans to snap pi3d into snapcraft https://snapcraft.io/search?q=pi3d ?

paddywwoof, 2019-05-23:
Hi, I've not tried core/snap but I'm happy to have a look. pi3d doesn't really have many dependencies as it's almost all vanilla python (2 or 3). It needs numpy (and PIL if  you want to do anything with image files or fonts i.e. 99% or use cases) and it needs to have OpenGLES shared object files.

It is quite usable on other non raspberry systems. On linux it uses mesa utils to emulate the OpenGLES and on windows it uses ANGLE and pygame. There is some explanation on the docs http://pi3d.github.io/html/ReadMe.html#setup-on-desktop-and-laptop-machines

matteoraggi, 2019-05-23:
just an image slider with crossfading effect for now :-) And I would like to create a fade in and fade out effects, like a crossfading to and from black slides too. 
Now that you gave me more info  about the fonts, maybe one day in the future also I will used pi3d also to show a  counter on a layer upper the image slider.

paddywwoof, 2019-05-23:
I'm actually in the process of making a simplified pictureframe app like the one used by Wolfgang https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/ also looking at being able to use large screens and I did put in an example of using the point text system. I will try and get it sorted and push the new version soon.

matteoraggi, 2019-05-23:
large screens = 1920x1080 ? or also bigger?

paddywwoof, 2019-05-23:
I wanted to do 4k but struggled on the RPi for some reason, possibly memory. It needs to have pi3d running in multiple process and pass info between (as per https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=163188&p=1056217&hilit=2048+screen+pi3d#p1056940) So I know it will run at screen sizes well over 1920.

I really need to get the latest bits of code sorted out and posted.

matteoraggi, 2019-05-23:
Normally Raspberry Pi 3 support max 1920x1080 videos, your is a great news!

paddywwoof, 2019-05-23:
Matteo, Not sure about the file load times on RPi3. The bigger the image the longer to load, but a second or so might be ok. Transition can be as quick as you want - RPi3 will run 40fps on simple graphics like this. Using file name for transition is quite easy in python. Something like

```
file_list = ["5-abcd.png","5-defg.png"]
for f in file_list:
   dwell_tm = int(f.split("-")[0])
```

matteoraggi, 2019-05-23:
Great! I am not a developer, would you like to help me?

paddywwoof, 2019-05-24:
Hopefully some of the improvements in the next release and related examples will be directly relevant.

matteoraggi, 2019-05-27:
GReat! do you have a ETA (Estimated time of arrival)? Or could I help yo uto test the alfa or beta version?

closed 2019-08-19 by paddywwoof