tipam/pi3d #236: Calling update_ndarray affects other textures in the same buffer

jean-emmanuel, 2019-06-11:
I'm using the 2nd texture slot of a buffer as a mask for the main texture in the 1st slot. The masking is done in a shader and everything works fine but if I call `update_ndarray(new_texture_array)` on the main texture, the mask texture gets discarded.
The only fix I found is to call `opengles.glBindTexture(GL_TEXTURE_2D, self.buf[0].textures[1]._tex)` after updating the main texture (calling `update_ndarray(None)` on the mask texture works too but increases cpu usage significantly for some reason).

Could it be that there is something in `update_ndarray` ?

paddywwoof, 2019-06-11:
yes could well be the case! As I said I can't look at this in detail till thursday, could you post a minimal bit of code that shows the kind of thing that goes wrong?Sent from my Samsung device

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/tipam/pi3d/issues/236?email_source=notifications\u0026email_token=AAOZPRUTAZMTBRN2NFO4JWTPZ63FBA5CNFSM4HW7EDK2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4GY2WT7A",
"url": "https://github.com/tipam/pi3d/issues/236?email_source=notifications\u0026email_token=AAOZPRUTAZMTBRN2NFO4JWTPZ63FBA5CNFSM4HW7EDK2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4GY2WT7A",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

paddywwoof, 2019-06-11:
there seem two oddities here. update_ndarray is for the vertex, normal, uv arrays not related, in theory, to the texture_2d sampler system so you might not need to use it unless you are changing vertex buffer stuff... but there does sound to be something getting reset in the call which needs fixing. If you call update_ndarray with no argument it uses the existing buffer entirely. If you pass an ndarray then that will be used as the arraybuffer, a texture  array will probably be bigger than the existing arraybuffer and will possibly overwrite things in an unpredictable way...

jean-emmanuel, 2019-06-11:
I'm passing a numpy array to update_ndarray, as in the [VideoWalk](https://github.com/pi3d/pi3d_demos/blob/master/VideoWalk.py) demo, the numpy array always has the same size, should I use a different method to update the texture ?

The aforementionned solution (`opengles.glBindTexture`) is actually not enough and does not work everytime. I'll make a minimal test case asap.

paddywwoof, 2019-06-11:
Hmm, sounds sensible on your part. I can't really remember how or why videowalk does what it does and can't really check it out from here (bunkhouse in Ullapool) on my phone. Will have a good look on Thurs night. It's probably a bug that needs squashing!

paddywwoof, 2019-06-11:
had a quick look on my phone and realize my first answer was way off as had forgotten update_ndarray was a method of Texture. I wonder if there is some assumption made in Buffer.draw() that causes the issue.. maybe the condition on line 300 or similar. Hopefully can wait a couple of days.

jean-emmanuel, 2019-06-11:
It can wait :)

Thanks for the hint, I'll tinker with Buffer.draw()

jean-emmanuel, 2019-06-12:
I found it ! All I need is to call `opengles.glActiveTexture(GL_TEXTURE0)` before `update_ndarray(ndarray)` to ensure `tex0` gets binded, otherwise it is the latest selected texture (by Buffer.draw()@[L302](https://github.com/tipam/pi3d/blob/master/pi3d/Buffer.py#L302)) that's affected.

I don't think there's anything wrong in Buffer.draw(), the problem is that `update_ndarray` assumes that `tex0` is always the active texture.

paddywwoof, 2019-06-13:
hi, thanks for digging this out. I will fix update_ndarray - prob shouldnt make that assumption.Sent from my Samsung device

closed 2019-06-27 by jean-emmanuel