tipam/pi3d #238: Not working on Raspberry Pi 4 (Raspbian)

m4r35n357, 2019-08-15:
Well I have tried some reliably working code on the Pi4 (Raspbian) and it certainly does not work for me.

In fact the test code in the README fails in the same manner (freezes silently) as my code. CTRL-C does not kill it, you need to do CTRL-Z and kill the job.

(ad) pi@raspberrypi:~/projects/ODE-Playground $ python --version
Python 3.7.3
(ad) pi@raspberrypi:~/projects/ODE-Playground $ pip list
Package         Version
--------------- -------
cycler          0.10.0 
kiwisolver      1.1.0  
matplotlib      3.1.1  
numpy           1.17.0 
pi3d            2.32   
Pillow          6.1.0  
pip             19.2.2 
pkg-resources   0.0.0  
pyparsing       2.4.2  
python-dateutil 2.8.0  
setuptools      41.0.1 
six             1.12.0 
wheel           0.33.4

paddywwoof, 2019-08-15:
Ian, I'll try and get that sorted right away. Just to check: are you running from the command-line with xinit or from the raspbian desktop? What have you set the OpenGL driver to? (i.e. it has been swapped to Fake KMS) Does it run at all and stop after a certain period or does nothing happen or does a new x11 window flash up then the console freezes, etc?

I don't see how the other apps would interfere but maybe I should have a look.

paddywwoof, 2019-08-15:
P.S. probably not relevant but as your code looks pretty much python3 only, is there anything different using
`pip3 list`

paddywwoof, 2019-08-15:
Ian, I've just tried the minimal demo from the Readme and it does run but because there is no keyboard detection or screen dimensions it defaults to full screen and then can't be closed (I use the 'windows' button to get the desktop home menu -> terminal -> top -> kill the process, not ideal. will patch this somehow)

Is this what you mean? I haven't spent the time to figure out what to do with your app but I increased display x and y so it doesn't make the window full screen then when I run
```
pi@raspberrypi:~/ODE-Playground $ python3 plotPi3d.py
1 2 3 4
```
I  get a blank screen. As I add more lines
```
pi@raspberrypi:~/ODE-Playground $ python3 plotPi3d.py
1 2 3 4
2 3 4 5
4 1 2 3
5 2 3 4
3 4 1 2
4 5 2 3
2 3 4 1
3 4 5 2
1 2 3 4
```
I get a little line drawn. However the mouse and keyboard input is stopped from working by the blocking input lines. How is the program supposed to be used?

m4r35n357, 2019-08-15:
Thanks for replying, I have a lot of things to look at tomorrow!  I will reply when I've digested your answers.  Please bear with me.
But so far you have reminded me there might be an issue with (something to do with) KMS on the Pi4.

m4r35n357, 2019-08-16:
I have just tried something and it looks like you might have nailed it with the KMS remark.  I currently have an issue with the Pi4 and have to disable kernel KMS (I haven't been keeping up with Pi technology to that degree!).
Basically, I have commented out this line in the PI4 part of the config file:
`#dtoverlay=vc4-fkms-v3d`
I did this because the default config is too big for my physical screen and I have to use the menu bar blind ;)  i.e the overscan settings only take effect if I disable KMS as above.
_If I restore that line the ball example in the README runs!_ (BTW I can't kill it, is there a key sequence for that?).
Now I don't really understand the implications of no KMS, "fake" KMS or "real" KMS, nor do I really know how to select or use each, but I suspect you might!
I leave it to your expertise to decide whether my config should be a supported mode of operation.  If not, I guess the issue should be closed.

m4r35n357, 2019-08-16:
> I get a little line drawn. However the mouse and keyboard input is stopped from working by the blocking input lines. How is the program supposed to be used? I would expect the input stuff to be in a different thread or use some non-blocking system.

Are you attempting to run my actual code?  If so cool, but you might find the following invocation more interesting ;)

`./tsm.py lorenz 9 10 .01 3000 -15.8 -17.48 35.64 10 28 8 3 | ./plotPi3d.py`

If not, pardon me.  To answer your question the code reads in lines either from standard input, or a data file.  The data is created by tsm.py, which is the "zoo" of ODEs.

paddywwoof, 2019-08-16:
Ian, Yes I was attempting to run your code! And, OK I will try your tsm lorenz suggestion, and see what the effect is of piping stuff into pi3d - I've never tried that so it will be very educational for me!

There must be some way of setting the screen size that doesn't require disabling the Fake KMS GL driver. I will have a search around.

m4r35n357, 2019-08-16:
Thanks for the feedback, I am quite proud of the ODE code!  BTW all the pi3d mouse/kb stuff is from the examples if that is what you mean.
Apparently the problem with pi3d is that without KMS I am just using a framebuffer, so OpenGL doesn't work (glxgears seem to work fine though - ?????!!!!), but the KMS driver ignores overscan settings so I can't use the desktop properly.
Also apparently the Raspbian people are awaiting some fix for this from "upstream", but I can't tell when or if that will happen.  In the meantime my RPi4 is a bit of a brick . . . :(

m4r35n357, 2019-08-16:
> P.S. probably not relevant but as your code looks pretty much python3 only, is there anything different using
> `pip3 list`
> 
> PPS though I see you get 3.7.3 in answer to
> `python --version`
> 
> are you running in a virtual env?
Yes, and pip3 gives the same answer as pip (as I expected).  My venv setup is described in the ODE-Playground README if you need to know more, but I don't think that is the problem now.

paddywwoof, 2019-08-16:
At one stage I got pi3d working on the Pi4 without the accelerated GL driver but it was unusably slow so opted for the present system of checking for accelerated drivers [see here](https://github.com/tipam/pi3d/blob/master/pi3d/constants/__init__.py#L86)

With the mouse and keyboard input I mean that they are expecting events to be read from the x11 server and accumulated in buffers to be read each display loop. If there is a python input statement in the display loop then it will stop until it gets a return character. This is what happened when I tried running plotpi3d directly. But something more complicated happens when the script is started by another process and inputs piped to it, it manages to keep it's own character readings at the same time as getting input from stdin.

Annoying that the screen doesn't fit and there's no way to force dimensions.

m4r35n357, 2019-08-16:
Ah OK now I understand what you meant ;)

BTW I am reading lines, not characters.  Is that enough of a difference, or am I continuing to miss the point?

Anyhow, really appreciate you helping me out here.  I think I will have to wait and see if the upstream "fix" ever appears.  Cheers.

closed 2019-10-27 by paddywwoof