tipam/pi3d #239: OffscreenTexture with alpha channel

jean-emmanuel, 2019-10-22:
Hi ! Do you know a way to make OffscreenTexture have an alpha channel ?

jean-emmanuel, 2019-10-22:
Sorry I just found out all I needed was to set the display's background transparent preserve transparency when drawing on the offscreen texture.

closed 2019-10-22 by jean-emmanuel

paddywwoof, 2019-10-26:
Hi @jean-emmanuel I've been away for a week in The Highlands of Scotland, not just no internet; no landline, mobile, TV, radio. Great actually! Glad you sorted out the alpha - I remember struggling to retain the z depth of the offscreen texture which makes the pi3d blur rather convoluted but I've not looked at the alpha.

Hope you're getting pi3d to do what  you need.

Paddy

jean-emmanuel, 2019-10-29:
We're running two instances of our [pi3d-based](https://github.com/PlagiatBros/pytaVSL) app on the same system to handle the VJing part of our show, it's working great :).