tipam/pi3d #240: Display.create ignores tk on linux

SnowyCoder, 2019-10-27:
Hello, I'm developing a software that I intend to use on a raspberry pi but I'm using my main linux machine for testing purposes. Reading the documentation I saw that the parameter "tk" in `Display.create` has the following description: "Do we use the tk windowing system?". I expected the parameter to create a tkinter window, but reading into the code this only happens in android and on a raspberry, in the rest of the platforms (ex. linux) the library ignores it and sets up a fake tk window. What's the reason behind this? Can you write the full behavior in the documentation?

paddywwoof, 2019-10-27:
Hi, Yes the tk stuff is a little confusing. On the Raspberry Pi (prior to v4) when using the bcm driver option (i.e. non-fakeKMS) the pi3d display layer isn't really 'attached' to the tk window which is sized and positioned to fit underneath it. I don't think any of the normal visual tk things work, such as buttons or other widgets work, just the mouse, key and event loop. If you want to use pi3d with tk (or gtk, qt etc) in a fully functional form then you probably have to render off screen then draw the result as an image as in the pi3d_demos examples GtkCube.py and PyQtCub.py

When I first tried to get pi3d working on other systems I looked at passing the drawing surface to EGL but didn't manage to get it to work, whereas using  x11 directly did seem to. Let me know why you want to use tk with pi3d and what RPi version you are targetting. I might be able to suggest ways round the problem?

SnowyCoder, 2019-10-28:
I was trying to go fullscreen and to hide the mouse pointer, my first idea to do it was to use a tk window so that I could change that. Now that I've tried pi3d on a real raspberry pi 3 I confirm that tk doesn't work and I'm searching a solution using bcm

paddywwoof, 2019-10-28:
It is possible to hide the cursor on an x11 window because that's what the sdl2 relativemousemode does and I use sdl2 for the window stuff in the rust_pi3d version of the code. However I'm struggling to get the xlib.XUndefineCursor() to work. Maybe I should look at sdl2 code to see what it's doing.

paddywwoof, 2019-10-28:
OK this seems to work. I will tidy it up and test it then put it in util/DisplayOpenGL.py so it fits in with the other display config stuff. On the subject of which you know you can set the window to full screen width by setting x=0,y=0 and no width or height or by specifying display_config constants in the arguments to Display.create().

```
      black = xlib.XColor()
      black.red = 0
      black.green = 0
      black.blue = 0
      noData = ctypes.c_char_p(bytes([0, 0, 0, 0, 0, 0, 0, 0]))

      bitmapNoData = xlib.XCreateBitmapFromData(self.d, self.window, noData, 8, 8)
      invisibleCursor = xlib.XCreatePixmapCursor(self.d, bitmapNoData, bitmapNoData, 
                                            black, black, 0, 0)
      xlib.XDefineCursor(self.d, self.window, invisibleCursor)
```

SnowyCoder, 2019-10-29:
Thanks for the help, I didn't know it could be this complicated with all those drivers and backends!

closed 2019-10-29 by SnowyCoder

paddywwoof, 2019-10-29:
In some ways the 'standardisation' of the RPi4 to use the mesa fake kms driver lost some of the nice features of running pi3d on the RPi but it does mean the X11 stuff is going to be the same from laptop to RPi. Running X11 driver code from python is a bit of a nightmare!

Anyway I will patch this as soon as I can and will add a bit more info about tk to the documentation.

Thanks for help.

SnowyCoder, 2019-10-29:
Thank you for the fast response, can I ask you a quick question? why don't you use a higher level library like glut or glfw?

paddywwoof, 2019-10-29:
When Tim started work on pi3d most of the higher level libraries wouldn't run on the RPi also the original model B and A took quite a bit of coaxing to get reasonable frame rates so we thought it would be better not to wrap too many layers of processing inside each other.

When I was trying to get pi3d working with the fake kms driver on the RPi4 (I had a problem rendering points) I did make a minimal program showing the issue with pyopengl and pysld2 and at some point I might make a full branch of pi3d using that technology and see how it performs - It would certainly simplify things and make running of different platforms easier.