tipam/pi3d #242: problems running on fedora28

3ndym10n, 2019-12-12:
Hello,
I'm trying to run some pi3d demos on my arm platform EAIDK-610, but

paddywwoof, 2019-12-12:
Hi, I think it has worked in the past on Mali GPU but can't remember what was involved; I will have a look back to see if I can find anything. Are you running with X11 server?

3ndym10n, 2019-12-13:
Thanks a lot. I use Fedora LXDE Desktop.

paddywwoof, 2019-12-13:
This https://groups.google.com/forum/#!topic/pi3d/b_JKtQ9e3to was the discussion with peter hess about using pi3d on mali gpu systems - I'm not sure if you can get much info from it.
I notice that the lib...so files are in /usr/lib64/ which may be significant as I struggled (without success) to get pi3d to work with the gentoo64 OS on the Raspberry Pi. While testing that I used the script below to see what configs were available with EGL. The problem was that none of the surface types had SURFACE_TYPE matching WINDOW_BIT (i.e. && 4).

It would be interesting if you could run the code below and let me know the result. Also it would be helpful to know precisely which demos crashed at line 130 and which crashed at line 102 i.e. is it just the use_glx argument to Display.create()?

It would be good to get to the bottom of this issue so any testing you can do would be very much appreciated.

``` python
from ctypes import (c_int, byref, Structure, CDLL, POINTER)
from ctypes.util import find_library

egl_name = find_library("EGL")
egl = CDLL(egl_name)

EGL_BUFFER_SIZE = 0x3020
EGL_ALPHA_SIZE = 0x3021
EGL_BLUE_SIZE = 0x3022
EGL_GREEN_SIZE = 0x3023
EGL_RED_SIZE = 0x3024
EGL_DEPTH_SIZE = 0x3025
EGL_STENCIL_SIZE = 0x3026
EGL_SAMPLES = 0x3031
EGL_SURFACE_TYPE = 0x3033
EGL_DEFAULT_DISPLAY = 0

EGLint = c_int

class _EGLDisplay(Structure): ###### EDIT needs this definition to work on gentoo64 is that significant (not needed on laptop 64)
    __slots__ = [
    ]
_EGLDisplay._fields_ = [
    ('_opaque_struct', EGLint)
]
EGLDisplay = POINTER(_EGLDisplay) #######

class _EGLConfig(Structure):
    __slots__ = [
    ]
_EGLConfig._fields_ = [
    ('_opaque_struct', EGLint)
]
EGLConfig = POINTER(_EGLConfig)

egl.eglGetDisplay.restype = EGLDisplay ####### EDIT needs this on gentoo64

display = egl.eglGetDisplay(EGL_DEFAULT_DISPLAY)
r = egl.eglInitialize(display, None, None)
attrib_dict = {EGL_RED_SIZE:"RED_SIZE",
			   EGL_GREEN_SIZE:"GREEN_SIZE",
			   EGL_BLUE_SIZE:"BLUE_SIZE",
			   EGL_DEPTH_SIZE:"DEPTH_SIZE",
			   EGL_ALPHA_SIZE:"ALPHA_SIZE",
			   EGL_BUFFER_SIZE:"BUFFER_SIZE",
			   EGL_SAMPLES:"SAMPLES",
			   EGL_STENCIL_SIZE:"STENCIL_SIZE",
			   EGL_SURFACE_TYPE:"SURFACE_TYPE"}
numconfig = EGLint(0)
poss_configs = (EGLConfig * 50)(*(EGLConfig() for _ in range(50)))
r = egl.eglGetConfigs(display, byref(poss_configs), EGLint(len(poss_configs)), byref(numconfig))
attr_val = EGLint()
for i in range(numconfig.value):
  print(i, "- ", end="")
  for attr in attrib_dict:
    r = egl.eglGetConfigAttrib(display, poss_configs[i], attr, byref(attr_val))
    print("{}={}, ".format(attrib_dict[attr], attr_val.value), end="")
  print("")
```

3ndym10n, 2019-12-16:
confused...i ran the test program you provided, and no result came out...I remember is it also the reason for the error in line 130???
I also retested the RunTests.py and found 3 demos in RunTests.py cashed at line 130:
"Minimal.py","SpriteBalls.py"and"Blur.py"
and the rest crashed at line 102...

paddywwoof, 2019-12-16:
Hi, I've edited the listing with a few more print()s to see where things are going awry. Could you give that a try.

Yes, the non-listing of EGL configs and the error on line 130 are probably very much related. line 102 is an assertion that there a context has been created and the context creation function (eglCreateContext) needs to be provided with a config.

The three demos that crash at line 130 are the ones that ask for GLX to be used for the creation of the window, so the program skips 72 to 103. But the error is just the first occasion that a GL library function (glGetString) returns something, and it turns out to be a null pointer. Presumably because something else in the configuration has silently failed to work.

Let me know what you find

3ndym10n, 2019-12-17:
Thanks for your help~openGL is too hard for me......I retried it today and here's the output:
```
1. libEGL.so.1.1.0
2. <CDLL 'libEGL.so.1.1.0', handle 557725f820 at 0x7f961e3898>
3. 0
4. 0
5. 0
```
Then I noticed that these files:
```
"libEGL.so"
"libEGL.so.1"
"libEGL.so.1.1.0"
"libGLESv2.so"
"libGLESv2.so.2"
"libGLESv2.so.2.1.0"
```
They all target to "libmali.so.r18p0".
And I also found another libEGL file named"libEGL_mesa.so.0.0.0" in /usr/lib64, so I edited line 4 into `egl_name = "libEGL_mesa.so.0.0.0"`, and here's what I got:
```
1. libEGL_mesa.so.0.0.0
2. <CDLL 'libEGL_mesa.so.0.0.0', handle 558fe26650 at 0x7f936cd6d8>
libEGL warning: DRI2: failed to authenticate
3. 1
4. 1
5. 20
0 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=0, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
1 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=16, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
2 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
3 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=8, SURFACE_TYPE=1031, 
4 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=32, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
5 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=0, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
6 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=16, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
7 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
8 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=8, SURFACE_TYPE=1031, 
9 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=32, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
10 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=0, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
11 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=16, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
12 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
13 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=8, SURFACE_TYPE=1031, 
14 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=32, ALPHA_SIZE=8, BUFFER_SIZE=32, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
15 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=0, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
16 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=16, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
17 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
18 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=24, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=8, SURFACE_TYPE=1031, 
19 - RED_SIZE=8, GREEN_SIZE=8, BLUE_SIZE=8, DEPTH_SIZE=32, ALPHA_SIZE=0, BUFFER_SIZE=24, SAMPLES=0, STENCIL_SIZE=0, SURFACE_TYPE=1031, 
```
I suppose it could be the "right" library???I don't know, but I changed that 3 libEGL.so* files' target file into "libEGL_mesa.so.0.0.0". Then I retried the demos, and found that all the domes used to crash at line 102 now crash at line 130 as well:
```
['Clouds3d', 5] >->
libEGL warning: DRI2: failed to authenticate
Traceback (most recent call last):
  File "Clouds3d.py", line 29, in <module>
    DISPLAY = pi3d.Display.create(x=MARGIN, y=MARGIN, frames_per_second=30)
  File "/usr/local/lib/python3.6/site-packages/pi3d/Display.py", line 563, in create
    display_config=display_config, window_title=window_title, use_glx=use_glx)
  File "/usr/local/lib/python3.6/site-packages/pi3d/util/DisplayOpenGL.py", line 130, in create_display
    if b"ES" in version:
TypeError: argument of type 'NoneType' is not iterable
```
emm......Is it because there's something wrong with my mali library? Should I try other libGLES libraries?

paddywwoof, 2019-12-17:
The problem looks similar to those I found trying to get pi3d working with gentoo64 on RaspryPi, so I don't think it's the GL library but the eglGetConfigs is finding surface types 1031 which do have third bit set so are suitable.

It would be good track down exactly which line gives the libEGL warning.That's probably key.

paddywwoof, 2019-12-17:
But, having said that... It would make sense to use the mesa library for GL if swapping for EGL. So do try that.