tipam/pi3d #243: position a gui element in corner of windows - track resizes?

pootle, 2019-12-18:
I want to position a couple of buttons in the window corners, and reposition them if the user resizes the window, but I can't see how to find the window's size, or detect when it is resized.

paddywwoof, 2019-12-18:
Good point. I see there's a TODO in DisplayOpenGL on that subject. It's probably rather difficult to do in your code as it needs an additional event mask setting up in DisplayOpenGL and the event checking in Display needs to catch resizes. i.e. DisplayOpenGL
``` python
      xlib.XSelectInput(self.d, self.window, KeyPressMask | KeyReleaseMask | ResizeRedirectMask)
```
and Display
``` python
      n = xlib.XEventsQueued(self.opengl.d, xlib.QueuedAfterFlush)
      for _ in range(n):
          xlib.XNextEvent(self.opengl.d, self.ev)
          if self.ev.type == KeyPress or self.ev.type == KeyRelease:
              self.event_list.append(self.ev)
          elif self.ev.type == ClientMessage:
            if (self.ev.xclient.data.l[0] == self.opengl.WM_DELETE_WINDOW.value):
              self.destroy()
          elif self.ev.type == ResizeRequest:
            (self.width, self.height) = (self.ev.xresizerequest.width,
                                         self.ev.xresizerequest.height)
            opengles.glViewport(0, 0, self.width, self.height)
            self.was_resized = True
```
then in programs
``` python
    if DISPLAY.was_resized:
        # move shapes and fix camera

```
However the the normal Camera.reset() with different lens settings doesn't work properly and creating new Cameras and setting `Shape._camera = new_cam` also has issues. I will let you know when I've figured out what to do, and will post the revisions to pi3d develop branch. Hopefully something quick to do.

Paddy

pootle, 2019-12-18:
Many thanks, that'll keep me going