tipam/pi3d #244: "X11 needs to be running" on Ubuntu 19.10 (intel cpu/gpu)

frenchfaso, 2020-03-01:
Hy there,
I'm having a hard time setting up pi3d on my intel laptop with Ubuntu 19.10.
I've installed the "mesa-utils-extra" package (also tried manually installing libgles2-mesa and libgles2-mesa-dev libs) and created a python3 venv with Pillow, Numpy and pi3d.
pi3d module imports just fine, but when I try to create a display with
`display = pi3d.Display.create()`
I get the following error:
> X11 needs to be running
> ...
> assert False, 'Couldnt open DISPLAY {}'.format(display_name)
> AssertionError: Couldnt open DISPLAY b':1024

..starting my app.py from ubuntu gnome-terminal.

paddywwoof, 2020-03-06:
Hi sorry for the delay responding. That error comes quite early on in the setup process before it even attempts to do any libgles2 stuff with the mesa drivers. There is a bit of code added last year to check the display number (as the Raspberry Pi started supporting multiple displays) and that looks to be doing something odd. What do you get if you type in a terminal:
`ls /tmp/.X11-unix`
I just get `X0` on this ubuntu (18.04) laptop. The next thing you could try that might give useful info, is in a python3 terminal:
>>> import ctypes
>>> xlib = ctypes.CDLL("libX11.so.6")
>>> xlib #should show you've loaded a DLL with a handle
>>> disp = xlib.XOpenDisplay(b":0") #that's the 0 from X0 showing displays
>>> disp #should give a non-zero number

frenchfaso, 2020-03-06:
Hy @paddywwoof thanks for your help!
ls -hl /tmp/.X11-unix gives me:
>total 0
>srwxrwxrwx 1 frenchfaso frenchfaso 0 mar  6 15:24 X0
>srwxr-xr-x 1 gdm        gdm        0 mar  6 15:24 X1024

..and here's the output of the interactive python session:
>\>>> import ctypes
>\>>> xlib = ctypes.CDLL("libX11.so.6")
>\>>> xlib
>\<CDLL 'libX11.so.6', handle 18c6350 at 0x7fa11ab56a90>
>\>>> disp = xlib.XOpenDisplay(b":0")
>\>>> disp
>25740784
>\>>> 

Seems like I have a second unix domain socket, X1024, which pi3d tries to use instead of X0.
I googled "X1024 unix domain socket" but didn't find anything useful.

paddywwoof, 2020-03-06:
Hmm, it's my fault probably. I just take the first entry from /tmp/.X11-unix. You could try a temporary hack.
[image]
either in the cloned pi3d repository if you're using that or /usr/local/lib/python.../dist-packages/pi3d/util/DisplayOpenGL.py
And see if that gets it working. Meanwhile I will see if I can work out a more scientific fix.

frenchfaso, 2020-03-06:
Thanks!
That did the trick :-D

closed 2020-03-06 by frenchfaso