tipam/pi3d #245: Name only shows for 1 second, and disappears.

Ascendin81, 2020-03-06:
Hello, I have pi3d running as a picture frame slideshow.  The database has over 100,000 photos, categorized by date in their name, and usually takes 5 hours to load via WiFi.

I enabled "show names", but I can not get the name to STICK for the entire life of the photo.  Can you please create the ability for us to show to show the name on TOP or SIDE of photo in a black bar with white text, so I can see when the photo was taken?

Thank you!

paddywwoof, 2020-03-06:
Hi, that's not too difficult to do but I appreciate it might be a tall order to become familiar with python then figure out how all the components of pi3d work!

The first thing I would suggest is that you copy a small  number of images to a folder then set your PIC_DIR to that to speed up the process of tweaking things the way you want them.

To stop the text fading you need to comment out the place where the alpha is gradually reduced:
[image]
Then, to give the text a black background you can either put a rectangle behind it, which is rather complicated, or just make the font have a coloured background. That's what I will show here, but the issue is that the character sprites overlap so they need to be spread out again, making the text look a bit 'sparse':
[image]
The position, size, rotation and justification can all be altered as in these different pictures. The location x=0,y=0 is in the middle of the screen with x increasing left to right, y increasing bottom to top. The width and height are attributes of DISPLAY. 
[image]
[image]
Let me know if you get stuck

closed 2020-03-06 by paddywwoof

Ascendin81, 2020-03-06:
Wow, Thank you!.  I will get this working today after work.  Amazing!