tipam/pi3d #247: How to add touchscreen support

Tsjippy, 2020-09-19:
Hi all, 

I tried to find it in the demo files but could not find it.
How can I setup touchsupport for pi3d?

I am currently using the PictureFrame2020.py on a RPI4 4GB with a Waveshare touchscreen.

I would like to be able to go to the next foto on touch

paddywwoof, 2020-09-21:
Hi, Sorry that the Mouse documentation is so non-existent  - it's generated automatically from the code docstrings using sphinx but somehow it doesn't pick anything up! I will try and fix that.

I'm not familiar with the Waveshare touchscreen but the first thing I would do is get that set up so it works OK with normal apps that require a mouse input (such as the desktop).

Then I would add mouse click checking to the Minimal.py demo just to check that worked. Something like
```python
...
mouse = pi3d.Mouse()
mouse.start()
last_button = mouse.button_status()
while DISPLAY.loop_running():
  sprite.draw()
  ...
  mb = mouse.button_status()
  if mb != mouse.BUTTON_UP and mb != last_button: # new click of some kind
    print(mb)
  last_button = mb
...
```
If that all worked as expected I would add the code to PictureFrame2020:
```python
...
if config.KEYBOARD:
  kbd = pi3d.Keyboard()
mouse = pi3d.Mouse() #put this near to the keyboard setup 
mouse.start()
last_button = mouse.button_status()
...
  text.draw()

  mb = mouse.button_status() # put this next to the keyboard checking code
  if mb != mouse.BUTTON_UP and mb != last_button:
    nexttm = time.time() - 86400.0 # use all the time up for the current slide so the next one starts
  last_button = mb

  if config.KEYBOARD:
    k = kbd.read()
```
You could make it behave differently depending on where you touch the screen (mouse.position() return x,y tuple)

Let me know how you get on

Paddy

Tsjippy, 2020-09-21:
thanks,

this really helps me!