tipam/pi3d #253: pi3d.Slice() unexpected angle

chatcoda, 2021-09-30:
Hello.
The following code
```
#!/usr/bin/python
import pi3d
DISPLAY = pi3d.Display.create(background=(255, 255, 255, 255))
slice = pi3d.Slice(inner=0.0, outer=0.3, name="slice", z=1, start=0, end=90)
mykeys = pi3d.Keyboard()

while DISPLAY.loop_running():
    slice.draw()
    k = mykeys.read()
    if k == 27:
        mykeys.close()
        DISPLAY.stop()

DISPLAY.destroy()
```
gives the following result:
[image]
Shouln't it render a 90° sector?
Thanks.

paddywwoof, 2021-10-01:
Thanks, that's a good catch. https://github.com/tipam/pi3d/blob/874b572682cf19812c29774939668676b3f1d4ca/pi3d/shape/Slice.py#L45 line is just wrong. It should be
```python
    inds = np.array([[2 * i + u, 2 * i + v, 2 * i + w] for i in range(0, self.n - 1)
                                    for (u, v, w) in [(0, 1, 3), (3, 2, 0)]], dtype=int)
```
In addition z doesn't seem to be set. I will update the repo on github and push a bugfix to pypi.org once tested

Paddy

paddywwoof, 2021-10-01:
updated on pypi https://pypi.org/project/pi3d/ so you should be able to `pip3 install --upgrade pi3d`

Let me know if there are any unexpected results.

Paddy

chatcoda, 2021-10-01:
I've just updated to 2.48 and it looks good now, thanks for the quick fix !
Here is a test I made to validate:
```
#!/usr/bin/python
import pi3d
DISPLAY = pi3d.Display.create(background=(255, 255, 255, 255))
slice = pi3d.Slice(inner=0.2, outer=0.3, name="slice", sides=360, z=1)
mykeys = pi3d.Keyboard()
end = 0

while DISPLAY.loop_running():
    end = (end + 1) % 360
    slice.reset
![pi3d_slice](https://user-images.githubusercontent.com/16852528/135644899-26c83b8f-21bd-44a8-8f15-5d441db0fbe8.gif)
_verts(end=end)
    slice.draw()
    k = mykeys.read()
    if k == 27:
        mykeys.close()
        DISPLAY.stop()

DISPLAY.destroy()
```
The result:
[image]

closed 2021-10-01 by chatcoda

paddywwoof, 2021-10-01:
Good.

Great minds think alike: I revised the Polygon.py demo that's run as part of RunTests.py for testing the library before release. It does a similar thing with non-zero inner and variable end!
https://github.com/pi3d/pi3d_demos/blob/8750ac5ae787973d50d2a21c557b1bde1fad858d/Polygon.py#L53