tipam/pi3d #255: gltf support?

TheBricktop, 2022-01-10:
Is there a possibility to integrate gltf/glb support into pi3d for model parsing? Since blender now has native gltf export with animations and even some materials (and textures) and whole industry adopting gltf as some sort of jpegs of 3d world, it is more and more common to use gltf as a standard model format.
I've found some python lib that may help with that:
https://github.com/sergkr/gltflib

paddywwoof, 2022-01-10:
Hi, that sounds like a good objective. I will look at that gltf spec as well, but the gltflib looks to be an excellent start.

I'm not sure how much pi3d will be able to encompass the full scope of gltf.  Animation doesn't really exist in pi3d beyond what you program things to do in python but in theory cameras, meshes, materials and textures could be imported though the mapping of meshes and materials to pi3d Shape and Buffer might require quite a lot of intervention.

TheBricktop, 2022-01-11:
That sounds like a lot yeah, I mean animations might be a bit overkill for starters ( especially bones and skinned meshes ) but I think that mesh and texture import would be enough to improve working with blender and allow importing free assets from sketchfab.