tipam/pi3d #256: problem running examples in CLI mode (no context)

TheBricktop, 2022-01-10:
tried to run the examples in CLI mode without the desktop rendering in the background for additional performance but ive run into a problem : 
[image]

paddywwoof, 2022-01-10:
Hi, that looks like the kind of error running with the "new" OpenGL mesa driver, which you have to do on RPi4 and later or with bullseye OS. In order for pi3d to get a surface to render onto it has to have an X11 driver running, though it doesn't need the desktop. The old broadcom dispmanx and OpenGLES system is faster on the lower spec RPis so it depends on what machine you are using. Note that bullseye seems to need glamour startomg in the config settings.

see http://pi3d.github.io/html/FAQ.html#running-from-command-line-with-the-new-fake-kms-mesa-v3d-vc6-etc
and https://github.com/helgeerbe/picframe/issues/195

TheBricktop, 2022-01-11:
I've tried to start using xinit but it seems like another x11 is runnning on the display i chose 
[image]