tipam/pi3d #258: Win 10 not every example is working

TheBricktop, 2022-01-13:
Tried the default installation on windows using provided dll's.
The Earth exampe runs correctly but almost every other doesnt.
Tried stereo example and the only output is light blue window.

for example :
```
from __future__ import absolute_import, division, print_function, unicode_literals

import math
import random
import pi3d

# Setup display and initialise pi3d
DISPLAY = pi3d.Display.create(w=1200, h=600, frames_per_second=30)
DISPLAY.set_background(0.4, 0.8, 0.8, 1)      # r,g,b,alpha
# yellowish directional light blueish ambient light
pi3d.Light(lightpos=(1, -1, -3), lightcol=(1.0,
           1.0, 0.8), lightamb=(0.25, 0.2, 0.3))
CAMERA = pi3d.StereoCam(separation=-0.5, interlace=0)

mykeys = pi3d.Keyboard()

while DISPLAY.loop_running():

    k = mykeys.read()
    if k > -1:
        if k == 112:
            pi3d.screenshot("earth1.jpg")
        elif k == 27:
            mykeys.close()
            DISPLAY.stop()
            break
```

gives me blue box

paddywwoof, 2022-01-13:
Hi, that's odd. It sounds like the mouse, or more probably keyboard input is stopping it running. The dlls are just OpenGLES functionality so if the `Earth.py` demo works that means the dll side is probably fine for all of them, but something else is causing the problem. On windows the mouse and keyboard input uses pygame, which is riding on SDL2.

I will have a go at using pi3d on a windows machine here (I haven't done that for a long time!)

TheBricktop, 2022-01-13:
Well Im only asking for that because im programming for both platforms and im probably going to teach pi3d for my students.

paddywwoof, 2022-01-13:
Well it used to work OK so it's something that needs fixing!

paddywwoof, 2022-01-13:
OK well I've just gone through the process of cloning pi3d, pi3d_demos and pi3d_windll then pip installing pygame, numpy and pillow, all 64 bit. And, remarkably, mostly it seems to work!

The ones that don't run at all seem to be a) ones using InputEvents but that's linux specific so has never worked on windows. b) Ones using offscreen textures i.e. blur, blur_legacy, foreststereo, all the filters also using `PostProcess` but the shadow ones work a bit, but not properly. The clash test does seem to work so it's probably some horrible little bug with a ctypes type or pointer. I vaguely remember having some problem with 64 bit previously and having to use the downloads from gohlke but not sure what the details were.

I will have a look at the code and see if I can see what the difference is between the `PostProcess`, `Defocus` and the `ShadowCatcher` , `ClashTest` they all inherit from `OffScreenTexture` but the first two just render the window with the background colour.

Do my results match yours in any way?

TheBricktop, 2022-01-13:
The example that has my concern is forestwalkstereo.py , while forestwalk works ok (there is some lag in the keyboard presses) the stereo version only renders background color. Also the dogfight crashes because of some problem with events .

paddywwoof, 2022-01-14:
Good. That sounds to be the same as I'm seeing. The demos using InputEvents such as Silo and Dogfight only need a couple of lines changing, but the problematic OffscreenTextures need a bug to be tracked down and fixed.

You are right, the keyboard input using pygame needs to be flushed every loop that it's read from. I will have a look at that too, it should be easy to fix.