tipam/pi3d #259: AttributeError: 'NoneType' object has no attribute 'glActiveTexture'

n21brown, 2022-01-18:
Rasp Pi 4 on Buster. Up to date, fresh install

When trying to run any of the demos I get 

pi@raspberrypi:~/pi3d_demos $ python3 Earth.py
Traceback (most recent call last):
  File "Earth.py", line 11, in <module>
    import pi3d
  File "/usr/local/lib/python3.7/dist-packages/pi3d/__init__.py", line 10, in <module>
    from pi3d.constants import (USE_PYGAME, PIL_OK,
  File "/usr/local/lib/python3.7/dist-packages/pi3d/constants/__init__.py", line 198, in <module>
    PLATFORM, bcm, openegl, opengles = _detect_platform_and_load_libraries()
  File "/usr/local/lib/python3.7/dist-packages/pi3d/constants/__init__.py", line 193, in _detect_platform_and_load_libraries
    set_gles_function_args(opengles) #function defined in constants/gl.py
  File "/usr/local/lib/python3.7/dist-packages/pi3d/constants/gl.py", line 387, in set_gles_function_args
    gles.glActiveTexture.argtypes = [GLenum] #
AttributeError: 'NoneType' object has no attribute 'glActiveTexture'
pi@raspberrypi:~/pi3d_demos $

This looks like the same error as before in [https://github.com/tipam/pi3d/issues/214](url)

paddywwoof, 2022-01-23:
Is this due to the type of driver set up in raspbi-config? I will see if I can reproduce this error.

paddywwoof, 2022-03-02:
Away at moment but will do some tests when I'm back. Also with 64 bit version

Jacques-Olivier-Farcy, 2022-03-27:
I have the same error on a fresh PI OS install and update with snake eyes from adafruit.

paddywwoof, 2022-03-28:
Thanks @Jacques-Olivier-Farcy,
In case anyone else comes across this thread - it seems to be related to the changes in RPi graphics driver. You can see discussion and solutions https://github.com/helgeerbe/picframe/issues/195#issuecomment-982999227

However the OP does say they are using original buster OS which hasn't loaded the OpenGLES library correctly. Not sure if this is fixed or not as I haven't reproduced it.

Jacques-Olivier-Farcy, 2022-03-28:
Hi there,
Yes i can confirm that with the Buster image of raspios and after a (long) update / upgrade, it is working fine.