tipam/pi3d #263: Windows Exception in DisplayOpenGL.py

Mettbrot, 2024-02-20:
I am trying to get to run pi3d via picframe on Windows. After installing all dependencies and putting the dll files in the directory, I still get this Exception in DisplayOpenGL.py

[image]

If I change this to catch None I get another error down the road because `Display.INSTANCE.opengl.max_texture_size` returns 0 which is obviously not correct.

Is there anything I am missing or is this usecase simply not supported on windows?

paddywwoof, 2024-02-20:
Hi @Mettbrot, I've not run pi3d on windows for a while (certainly not on this current laptop) and when I read your posts on the picframe repository I thought I had better check things out.

My feelings are that it would be much simpler to use the SDL2 wrapper on everything apart from the older Raspberry Pis. But, as I say, I haven't tried that yet. I will give it a go now and get back to you with my findings!

Paddy

paddywwoof, 2024-02-20:
Hmm, I've tried to patch a couple of places so it uses SDL2 instead of pygame but I still need the `libEGL`, `libGLESv2` as well as the `SDL2` dlls. I then run into the problem with the `opengles.glGetString(GL_VERSION`) returning `null` as you did, then the Texture loading not working. I get a display surface popping up for a second so I think the SDL2 side is working but there seems to be something wrong with the OpenGLES library. I will try and figure it out later.

Mettbrot, 2024-02-21:
Thank you for looking into this. Let me know if there is anything I can help you with.