tipam/pi3d #264: Uppercase keys no longer working

lkesteloot, 2025-01-25:
I was using v2.22 of pi3d (on RPi 2), and the keyboard's `read()` function was properly returning uppercase letters (shift-A was returning 65). I've upgraded to the latest version (v2.52) and all letters now return the lowercase value (97 for shift-A). The shift key itself returns 0 (as does other modifier keys). I tried `read_code()` and it's the same.

I checked the sample programs and they all only ever use lowercase commands.

This is on Linux on Raspberry Pi 5, console mode (via xinit). I tried with and without curses.

paddywwoof, 2025-01-25:
Hi @lkesteloot the keyboard and mouse functionality on the later Raspberry Pus definitely needs sorting out. What OS are you using, and are you using the SDL2 system. Sounds like you're using X rather than Wayland.

Definitely sounds like a bug if the modifier always return 0

Paddy

lkesteloot, 2025-01-25:
Hi!

My current setup:

- Raspberry Pi 5 Model B Rev 1.0
- Debian GNU/Linux 12 (bookworm), kernel 6.6.62
- I assume I'm not using SDL2 since it's forced to `False` in `Display.py` (line 115) on the Raspberry Pi.
- My desktop is set up to use Labwc with Wayland, but here I'm running it from the console with `xinit`.

With `use_curses` set to either `True` or `False`, all modifier keys return 0, and the `Q` key always returns 113, whether shifted or not. I don't see a way to detect uppercase letters.

(BTW the project is [PiSlide](https://github.com/lkesteloot/PiSlide), which uses uppercase letters for dangerous operations like quitting and enabling debugging mode.)

Thank you for responding! I'm happy to do whatever tests you'd like.

paddywwoof, 2025-01-26:
Hi @lkesteloot I've just had a quick look at this (pi3d.Keyboard) The curses keyboard will only available if pi3d thinks it's running on an older raspberry pi using the libbcm.so graphics driver rather than the newer mesa ones. https://github.com/tipam/pi3d/blob/df0cc84aebc56fb8d62e2517aeb48e59da689681/src/pi3d/Keyboard.py#L307 I'm not sure why the keyboard functionality would be related but I think it just ties in with older versions of the OS and what utilities have been deprecated. Certainly if I comment out the condition and force curses to be used on this RPi5 it doesn't work!!

I think this same issue came up in the past when running pi3d on different platforms and the way round it was to use the `X11Keyboard` (which will be the default on later Pis in the absence of SDL2) and keep track of which modifier characters had been pressed. This was the system we used in `pi3d_demos.Gui.py`. Could you try this test program to see it it works on your new Pi?
```python
import pi3d
         
DISPLAY = pi3d.Display.create(w=100, h=100, frames_per_second=30)
keyboard = pi3d.Keyboard(use_curses=True)
print(f"actual type of keyboard running here={type(keyboard)}")
shifted = False
caps = False

while DISPLAY.loop_running():
  kc = keyboard.read_code()
  if len(kc) > 0:
    is_char = (type(kc) == str and len(kc) == 1)
    if kc == chr(27) or kc == "Escape":
      break
    if "Shift" in kc:
      shifted = True
    elif "Caps" in kc:
      caps = not caps
    elif is_char:
      if shifted or caps:
        resultant_char = kc.upper()
      else:
        resultant_char = kc
      print(f"{kc} {ord(kc) if is_char else ''} resulting in {resultant_char}")
      shifted = False
    else:
      print(kc) # an non char key other than Shift.. or Caps..

DISPLAY.destroy()
```

paddywwoof, 2025-01-26:
For the `sdl2Keyboard` I need to do something like this in `__init__`
``` python
    self.KEYBOARD = {}
    for k in dir(sdl2): # sift through for key codes
      if "SDLK_" in k:
        key_num = getattr(sdl2, k)
        if type(key_num) == int:
          key_code = k.split("_")[-1]
          self.KEYBOARD[key_num] = key_code
...
def _update_event(self):
...
    self.key_code = self.KEYBOARD[k]
```
SDL has the extra info in the `Display.keys_pressed[k][1]` showing the key_mod status, which is held in the `Keyboard` instance and might get round some issues. However there are multiple keyboard events for each key press and the SDLK constants are different from the X11 ones!

lkesteloot, 2025-01-26:
I tried your test code from [this comment](https://github.com/tipam/pi3d/issues/264#issuecomment-2614341898). The type of the class is `x11Keyboard`.

It didn't show any keys at all! `read_code()` always returns an empty string, and `read()` always return -1. Took me a while to debug, but eventually figured out that the size of the display is too small, and instead of `w=100, h=100` you have to pass in `w=800, h=500`. Maybe it changed the timing so that both an X11 press and a release got enqueued, and `_update_event()` processed both, losing the press. (The program otherwise worked fine with 100x100, including drawing a sprite when I modified it to do that.)

Secondly, if the user holds Shift, then presses the A key twice, the first one is uppercase but the second is lowercase. The keyboard class doesn't pass on releases, so we can't know when the shift key is released.

Want me to give it a go to modify the `x11Keyboard` class to internally track modifier keys and return the correct ASCII key from `read()`? I'll also enqueue keys so that they never get lost between polls. (I don't know what `read_code()` is even supposed to return, so I won't mess with that one.)

lkesteloot, 2025-01-27:
Also note that `pi3d.PLATFORM` is 3, which is `PLATFORM_LINUX`, not `PLATFORM_PI` as I'd expected! I had written above that SDL2 is forced to False on Pi, which is true according to the code, but false in my case since pi3d thinks my platform is Linux.

paddywwoof, 2025-01-27:
Hi, that's a bit weird: the display dimensions disrupt the key reading!

As you can probably tell, the code has evolved in a rather nonlinear way as the hardware changed. PLATFORM_PI means early models, with various closed source Broadcom software required and a reasonable probability that no X11 server would be running. The aim of the RPi foundation has been to align themselves with standard Linux so software can be ported with the minimum of low level hacking. Hence later Pi models are identified as PLATFORM_LINUX

Also, yes, it would be *great* if you had the time and inclination to look at the Keyboard code.

rec, 2025-01-27:
> the display dimensions disrupt the key reading!

Just unlurking to say that this is a big surprise!

...and also, "I'm still around".  I'd help except I don't even have an RPi here, we recently moved to Rouen and our stuff is in storage in Amsterdam.

Thanks for all your work on this, Paddy, and come and visit some time!

lkesteloot, 2025-02-21:
Hi! Here's a program that works well. It uses xlib's `XKeycodeToKeysym()` and `XKeysymToString()` to get all the values we need. There was no spec for pi3d's keyboard's `read()` and `read_code()` methods, so my code does something reasonable for each and hopefully it won't break too much. All traditional keys (letters, upper and lower case, enter, backspace) that can be expressed in ASCII are returned as the ASCII number. Non-standard things like arrow keys return a number that's already being used by pi3d (e.g., 136 for left arrow). The strings returned by `read_code()` look great.

An important improvement is that there's now a queue to store all keys, so that they won't get lost if the user types too fast.

I don't have the source of pi3d checked out, so I'll leave it to you to integrate this into the `x11Keyboard` class. Shouldn't be too hard, just make `key_queue` a field and the others methods. I kept the check for `display` being `None` so that you can just copy the code as-is.

```py
import pi3d
from pyxlib import x
import pyxlib
import json

DISPLAY = pi3d.Display.create(w=800, h=500, frames_per_second=30)
keyboard = pi3d.Keyboard()
print(f"actual type of keyboard running here={type(keyboard)}")
print(f"platform {pi3d.PLATFORM}")

# Queue of (key number, key string) tuples.
key_queue = []

def read_key_and_string(keyboard):
    global key_queue
    if keyboard.display is None: #Because DummyTkWin Keyboard instance created before Display!
      from pi3d.Display import Display
      keyboard.display = Display.INSTANCE
    while len(keyboard.display.event_list) > 0:
        e = keyboard.display.event_list.pop(0)
        if e.type == x.KeyPress:
            keycode = e.xkey.keycode
            keysym = pyxlib.XKeycodeToKeysym(DISPLAY.opengl.d, keycode, e.xkey.state)
            if keysym < 128:
                # Letters, numbers, symbols. Handles modifier keys.
                key = keysym
                key_string = chr(keysym)
            else:
                key_string = pyxlib.XKeysymToString(keysym).decode()
                if keycode < len(keyboard.KEYBOARD):
                    # Other action keys (Return, Backspace, Tab, Esc).
                    key = keyboard.KEYBOARD[keycode][0]
                else:
                    # Modifier keys.
                    key = 0
            key_queue.append( (key, key_string) )

    return key_queue.pop(0) if key_queue else (-1, "")

def read(keyboard):
    return read_key_and_string(keyboard)[0]

def read_code(keyboard):
    return read_key_and_string(keyboard)[1]

while DISPLAY.loop_running():
    key, key_code = read_key_and_string(keyboard)
    if key >= 0:
        print("key", key, "key_code", key_code)
        if key == 27:
            DISPLAY.stop()

DISPLAY.destroy()
```

paddywwoof, 2025-02-21:
Lawrence thanks so much for working on this. I've started the process of harmonising the pi3d Mouse and Keyboard behaviour for SDL2, X11 etc so this is just what I need :)

paddywwoof, 2025-02-21:
Lawrence, I've updated pi3d develop branch https://github.com/tipam/pi3d/commit/09868403f4812e3e3b2a01b22c3349cc3cb4929b I used the existing `Display.event_list` as the key_queue, just stopped the method emptying it each time it's called!

`XKeycodeToKeysym` seems to be bit picky, if the state is anything but 1 or 2 it won't return anything. I've put in some checking.